ArduPilot/MAVProxy #97: Segmentation fault on yosemite

joula, 2015-01-08:
When running mavproxy.py I get this error:

root@mac:~/drone$ mavproxy.py
Auto-detected serial ports are:
Logging to mav.tlog
Segmentation fault: 11

The mav.tlog file is completely empty.

I have installed the latest python with homebrew.

Here are the detailed log:
Process:               Python [66844]
Path:                  /usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.9 (2.7.9)
Code Type:             X86-64 (Native)
Parent Process:        bash [582]
Responsible:           Terminal [577]
User ID:               501

Date/Time:             2015-01-07 23:52:27.970 +0100
OS Version:            Mac OS X 10.10.1 (14B25)
Report Version:        11
Anonymous UUID:        190A0D93-4698-C383-6EDA-CCF1019618AD

Sleep/Wake UUID:       E4B2E672-DA0A-4547-A382-C4C2B83DE07F

Time Awake Since Boot: 770000 seconds
Time Since Wake:       440 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000

VM Regions Near 0:
--> 
    __TEXT                 0000000100811000-0000000100813000 [    8K] r-x/rwx SM=COW  /usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	000000000000000000 0 + 0
1   org.python.python             	0x000000010190e0dd PyEval_GetGlobals + 23
2   org.python.python             	0x000000010191d62b PyImport_Import + 137
3   org.python.python             	0x000000010191bd27 PyImport_ImportModule + 31
4   cv2.so                        	0x000000010171b1d7 initcv2 + 26
5   org.python.python             	0x00000001008b32cf _PyImport_LoadDynamicModule + 140
6   org.python.python             	0x00000001008b2fb7 import_submodule + 270
7   org.python.python             	0x00000001008b2bc2 load_next + 345
8   org.python.python             	0x00000001008b0d89 PyImport_ImportModuleLevel + 1135
9   org.python.python             	0x00000001008963db builtin___import__ + 135
10  org.python.python             	0x00000001008210ea PyObject_Call + 99
11  org.python.python             	0x00000001008a0d8b PyEval_CallObjectWithKeywords + 93
12  org.python.python             	0x000000010089d065 PyEval_EvalFrameEx + 8375
13  org.python.python             	0x000000010089ad7a PyEval_EvalCodeEx + 1409
14  org.python.python             	0x000000010089a7f3 PyEval_EvalCode + 54
15  org.python.python             	0x00000001008af836 PyImport_ExecCodeModuleEx + 241
16  org.python.python             	0x00000001008b2516 load_source_module + 1091
17  org.python.python             	0x00000001008b2fb7 import_submodule + 270
18  org.python.python             	0x00000001008b2de3 ensure_fromlist + 330
19  org.python.python             	0x00000001008b0e49 PyImport_ImportModuleLevel + 1327
20  org.python.python             	0x00000001008963db builtin___import__ + 135
21  org.python.python             	0x00000001008210ea PyObject_Call + 99
22  org.python.python             	0x00000001008a0d8b PyEval_CallObjectWithKeywords + 93
23  org.python.python             	0x000000010089d065 PyEval_EvalFrameEx + 8375
24  org.python.python             	0x000000010089ad7a PyEval_EvalCodeEx + 1409
25  org.python.python             	0x000000010089a7f3 PyEval_EvalCode + 54
26  org.python.python             	0x00000001008af836 PyImport_ExecCodeModuleEx + 241
27  org.python.python             	0x00000001008b2516 load_source_module + 1091
28  org.python.python             	0x00000001008b27d7 load_package + 303
[... 349 lines cut ...]
__LINKEDIT                         74.2M
__TEXT                            135.4M
__UNICODE                           544K
shared memory                         4K
===========                      =======
TOTAL                             358.7M

Model: MacBookPro10,1, BootROM MBP101.00EE.B05, 4 processors, Intel Core i7, 2.4 GHz, 8 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe, 1024 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, 0x45424A3431554638424353302D474E2D4620
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, 0x45424A3431554638424353302D474E2D4620
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.15.124.12.10)
Bluetooth: Version 4.3.1f2 15015, 3 services, 19 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SD256E, 251 GB
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: FT231X USB UART
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4

joula, 2015-03-10:
Not relevant any more.

closed 2015-03-10 by joula

anujdeshpande, 2015-05-18:
@joula how did you get rid of this ?

joula, 2015-05-18:
@anujdeshpande I couldn't fix the problem on OSX but switched to Node.js library to interact with mavlink.