ArduPilot/MAVProxy #112: Can you exit MAVProxy cleanly from inside a module?

nickponline, 2015-03-04:
Hey guys, 

What is the recommended way to get MAVProxy to exit cleanly from inside a module?

gmorph, 2015-03-04:
As far as I know there is nothing particular you have to do.  Are you having a specific problem we can look at?  The joystick module unloads cleanly but doesn't have any particular code to handle anything.  The console module which does have a couple of lines doesn't unload cleanly.  But it doesn't stop MAVProxy running.
Thanks, Grant.

nickponline, 2015-03-06:
Nothing in particular, I'm currently signaling to the main thread to close the child threads and then exit from there so it's clean as opposed to just calling sys.exit in a module.