ArduPilot/MAVProxy #121: Raspberry Pi UDP Issue

jabelone, 2015-04-08:
I have gotten MAVProxy mostly working on my Raspberry Pi 2 running raspbian.  I have a pixhawk plugged in via USB that is automatically detected by MAVProxy on startup.  I have added the --out=192.168.0.5:14550 option to see if I can connect remotely.  Everything seems to work well untill I try to connect using any GCS.  As soon as I hit the connect button, this error pops up on the MAVProxy terminal window:

`<Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "build/bdist.linux-armv7l/egg/MAVProxy/modules/mavproxy_map/srtm.py", line 134, in createFileListHTTP
    r1 = conn.getresponse()
  File "/usr/lib/python2.7/httplib.py", line 1034, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 407, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 371, in _read_status
    raise BadStatusLine(line)
BadStatusLine: ''>`

gmorph, 2015-04-08:
I assume that 192.168.0.5 is a separate computer running a GCS like Mission Planner?  And on Mission Planner you select UDP and click connect?
I'm not sure what the error above is.  Running the latest MAVProxy is always a good idea.  And latest GCS software.
Have you had this successfully running this setup and a computer other then the RPi?  
Thanks, Grant.

jabelone, 2015-04-09:
Yep, that is exactly what I'm doing - I've tried it both on the latest Mission Planner and APM Planner.  These are the exact steps I am undertaking to reproduce the problem.

1) Plug pixhawk into rPi.
2) Run the following command `<sudo mavproxy.py --out=192.168.0.7:14550 --console>` (I changed the IP to a static one, hence the change from original post)
3) From win 8.1 laptop on same network with ip of 192.168.0.7, using mission planner or APM planner I click on UDP and connect.
4) Mission planner asks for the port (14550).  Once I click "Ok" on this window or leave it for a little while, the original error above is printed out on the MAVProxy terminal window that's running on the rPi.

jabelone, 2015-04-09:
UPDATE:  It is working fine now!

I just noticed that for some reason the rPi didn't appear to be communicating with the pixhawk.  Nothing was being output in the MAVProxy module console and none of the flight mode etc items wwere being populated. (don't ask me how I didn't notice this before - doh!)

I simply re-installed the arduplane firmware onto the pixhawk and it started working.  Was this just a really unlikely edge case?  Does MAVProxy assume that because it can see a flight controller is connected, it can talk to it?  Maybe it would be a good idea to add a "Flight controller detected but no heartbeat packets received." type error message to MAVProxy if this was the case.

gmorph, 2015-04-09:
Normally if MAVProxy can't communicate with the PH you won't see anything in the console.  At any rate glad its working now.  I'm going to close this issue but if it happens again we can raise another one and do further diagnostics.  Thanks.

closed 2015-04-09 by gmorph

jobatall, 2015-05-06:
Hi, I'm having the same problem... I installed the mavproxy with all the needed packages in the raspberry pi model b. When I plug the APM 2.6 via microUSB and the raspberry pi via usb. I tried ttyACM0 and ttyAMA0 with 115200 and 56700 baudrates and with all these possible combinations I get the same result always... I don't know how I can connect the APM to the raspberry... I tried to put lsusb to see if the raspberry recognize the apm and, yes, the raspberry knows that the apm is plugged in an usb port... 
So I don't know what else I can do to have a good connection... 

Thanks