ArduPilot/MAVProxy #141: GUI modules with wx are broken on OSX

mrpollo, 2015-05-06:
Trying to use MAVProxy with either console or map kills those module Threads and keeps MAVProxy is running.

Installed wx from homebrew ```brew install wxmac``` I can confirm wx is working.

Python Stack:
```
☁  flight_replay [tmp_flight_replay66] ⚡ mavproxy.py --master=127.0.0.1:14550 --console
Connect 127.0.0.1:14550 source_system=255
Logging to mav.tlog
Running script /Users/rroche/.mavinit.scr
-> module load droneapi.module.api
DroneAPI loaded
Loaded module droneapi.module.api
-> set moddebug 2
/usr/local/lib/python2.7/site-packages/MAVProxy/modules/lib/wxconsole.py:53: wxPyDeprecationWarning: Using deprecated class PySimpleApp.
  app = wx.PySimpleApp()
Loaded module console
MAV> Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/site-packages/MAVProxy/modules/lib/wxconsole.py", line 62, in watch_thread
    msg = self.parent_pipe_recv.recv()
EOFError
```

System Stack Trace: https://gist.github.com/mrpollo/7501b3a0a277498e98b0

What other details can I provide to help track down the root of this problems, I'm not sure where to look for a fix here, I can provide a PR if someone can help out understand the depth of the problem.

amilcarlucas, 2018-10-17:
Can you retry with master?

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas