ArduPilot/MAVProxy #142: cmd not seeming to work for wp load

afferent, 2015-05-08:
I have tried on 1.4.11(win), 1.3.13(win) and linux (latest) and I can't seem to get this to work.

The command I have tried is "--cmd="wp load [filename.txt];"

What happens is it loads up and says "x number of waypoints loaded" but when I check no new waypoints are there. It is as if it is trying before its connected. (UDP LINK --master=X.X.X.X:PORT)

Using the same commands in the CLI works and if i load them from a script in the CLI it works, but if I load a script it just says "x waypoints loaded".

Is there some trick to get "wp load" to work from cmd?

Thanks
picture shows pink arrow firstly clears out waypoints
red arrow tries to load them from cmd
green arrow works on the CLI (to prove it loads)

[image]

gmorph, 2015-05-08:
I replicated this bug.  I plugged a pixhawk on my bench into my linux laptop and ran
mavproxy.py --console --map --aircraft test
and did a "wp list" and there were no mission waypoints loaded.  I then ran
mavproxy.py --cmd="wp load /full directory path/waypoints/mission.wpt" --map --console --aircraft test
and then another "wp list" inside MAVProxy and it listed the waypoints no problem. 

Now I realise your trying to establish a network connection via MAVProxy.  So I did the same command above and then inside MAVProxy ran "output add 127.0.0.1:12345" and in a separate terminal ran a second mavproxy like:
mavproxy.py --map --console --master 127.0.0.1:12345 --cmd="wp load /full directory path/waypoints/mission.wpt"
And saw the same error you documented above.  It says it loaded the waypoints but then a "wp list" show's nothing.

So bug confirmed.

Thanks, Grant.

lfway, 2016-12-20:
Hi all, 
I have this problem too. Do you have found solution?
Thank you.

peterbarker, 2017-02-16:
`wp load` is unlikely to work when given on the command-line; in fact, nothing that requires you to be connected to the vehicle is likely to work, including things like mode changes.

mavlink is a lossy protocol; when you do these commands it is sending a message into the void and hoping for a response.  Since no vehicle is connected, the packet is lost.

Even from the command-line, a `wp load` is not guaranteed to work.  If the first packet is lost it will just fail silently (and we 've seen that happen on real aircraft).

The "loaded waypoints" message refers only to MAVProxy successfully parsing the supplied file.  We could add a "Sending waypoint count" message to the user, but that would be getting somewhat verbose, I think.

Sorry, but I don't think this is a bug?  Concur?