ArduPilot/MAVProxy #147: Hardcoded paths called

lvale, 2015-05-18:
Some commands and functions refer to hardcoded paths that are not present on a user machine (latest Win Version). As an example:

Loaded module console
MAV> ERROR in command: 'NoneType' object has no attribute 'waypoint_clear_all_se
nd'
Traceback (most recent call last):
  File "<string>", line 462, in process_stdin
  File "C:\PROGRA~2\MAVProxy\eggs\mavproxy-1.4.11-py2.7.egg\MAVProxy\modules\mav
proxy_wp.py", line 525, in cmd_wp
    self.master.waypoint_clear_all_send()
AttributeError: 'NoneType' object has no attribute 'waypoint_clear_all_send'
ERROR in command: 'NoneType' object has no attribute 'param_set_send'
Traceback (most recent call last):
  File "<string>", line 462, in process_stdin
  File "C:\PROGRA~2\MAVProxy\eggs\mavproxy-1.4.11-py2.7.egg\MAVProxy\modules\mav
proxy_fence.py", line 192, in cmd_fence
    self.param_set('FENCE_TOTAL', 0, 3)
  File "C:\PROGRA~2\MAVProxy\eggs\mavproxy-1.4.11-py2.7.egg\MAVProxy\modules\lib
\mp_module.py", line 102, in param_set
    self.mpstate.functions.param_set(name, value, retries)
  File "<string>", line 211, in param_set
  File "C:\Users\tridge\Documents\GitHub\MAVProxy\MAVProxy\build\mavproxy\out00-
PYZ.pyz\pymavlink.mavparm", line 22, in mavset
AttributeError: 'NoneType' object has no attribute 'param_set_send'

stephendade, 2015-05-20:
There's an updated version of MAVProxy at http://firmware.diydrones.com/Tools/MAVProxy/ that should help you.

lvale, 2015-05-20:
Downloaded the "latest" (I assume the latest is 2015-05-13) and similar issues 
[image]

stephendade, 2015-05-20:
I can't seem to be able to replicate the issue. Which flight controller (and version) and you connecting to?
Are you running MAXProxy with any commandline options?

lvale, 2015-05-20:
No FC attached. And also no cmdline options on the screenshot.

stephendade, 2015-05-20:
Any chance you could scroll to the top of your console so I could see the entire output? It should start with "Logging to mav.tlog"

lvale, 2015-05-20:
The console when opened, without any commands issued, yet. Still the eggs dir :)

[image]

lvale, 2015-05-20:
btw:

This is quite easy to duplicate. The screenshots are from a freshly installed Windows7-64. So a similar VM with a freshly installed (and Windows updated) replicates this behavior easily...

stephendade, 2015-05-21:
The "Using deprecated class PySimpleApp" message is a known warning. It doesn't affect the functionality of the program.
As for the "EOFError", that is normal when exiting MAVProxy.
How did you get the "Nonetype object has no attribute 'param_set_send' error from previously? I'm unable to replicate that particular message.

lvale, 2015-05-22:
If I remember correctly that came up because I chose Clear Mission without FC attached.

stephendade, 2015-05-22:
Ahh yes, I see now.

The issue is that some MAVProxy modules don't check to see if there's a valid connection before executing a command.

The hardcoded directories in the error messages are an artifact of the compilation process, but do not affect the running of the program.

lvale, 2015-05-22:
ok. thanks

closed 2015-05-22 by lvale