ArduPilot/MAVProxy #148: Add DroneKit-Python to the windows installer

hamishwillee, 2015-05-27:
This would make using DroneKit on windows much easier.

hamishwillee, 2015-05-27:
FYI According to Stephen Dade:

> In terms of compiling this into an easy-to-run exe, that is going to be far more difficult. Because dronekit-python uses a different directory structure when installed (via setup.py) that what the other MAVProxy modules do, the MAVProxy installation script won't find the dronekit files. There are two ways to solve this:
1. Patch dronekit to have the correct directory structure
2. Patch MAVProxy installation script to find dronekit

djnugent, 2015-06-04:
I just ran through the install instructions for Dronekit-python windows and I don't see the advantage of having dronekit install with mavproxy. Mavproxy installs as a dependency of dronekit anyway(so the they install at the same time).
 Adding it to mavproxy would remove one command, "echo module load droneapi.module.api >> %HOMEPATH%\AppData\Local\MAVProxy\mavinit.scr" 

Also the end goal is to remove mavproxy as a dependency of dronekit, so this would move us in the opposite direction.

hamishwillee, 2015-06-05:
@djnugent This is a "one click" windows installer package which installs MAVProxy on Windows with all its dependencies. The benefit of adding DroneKit to this build would be that our instructions for setting up DroneKit on Windows would turn into "install this exe". Much easier. 

In addition, while MAVProxy is a dependency of dronekit, you don't get ALL the MAVProxy dependencies (if you follow the current instructions you don't get the MAVProxy map or console). Now while we do need instructions to get DK and MAVProxy to set up perfectly in ActiveState Python (with all dependencies) this is still going to be much harder than instructions "press a button".

If we eventually remove the MAVProxy dependency we'd revisit :-)

Hope this makes sense! This is IMO the single best thing we could do to support Windows DK use right now.

hamishwillee, 2015-06-15:
Closed by https://github.com/diydrones/dronekit-python/pull/161 . THanks @stephendade .

closed 2015-06-15 by hamishwillee