ArduPilot/MAVProxy #167: Land mode command doesn't work

KiwiHC16, 2015-07-21:
Using SITL for some tests. I can set stabilize, loiter, auto,... from mavproxy but when is enter "land" then the copter stay in current mode and refuse to go to land mode. I have to use Mission Planner to set land mode. Bug ?

stephendade, 2015-07-22:
Yep, I can replicate this issue on my setup.
I believe that APM:Plane and APM:Copter implement the LAND mode selection slightly differently. Currently MAVProxy only supports APM:Plane using the "land" command.
It would be useful to have the shortcut working for APM:Copter too.

KiwiHC16, 2015-07-24:
In commun.xml from mavlink was have both definition:
                <entry value="189" name="MAV_CMD_DO_LAND_START">
                    <description>Mission command to perform a landing.This is used as a marker in a mission to tell the autopilot where a sequence of mission items that represents a landing starts. It may also be sent via a COMMAND_LONG to trigger a landing, in which case the nearest (geographically) landing sequence in the mission will be used. The Latitude/Longitude is optional, and may be set to 0/0 if not needed. If specified then it will be used to help find the closest landing sequence.</description>
                    <param index="1">Empty</param>
                    <param index="2">Empty</param>
                    <param index="3">Empty</param>
                    <param index="4">Empty</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Empty</param>
                </entry>
and 
             <entry value="176" name="MAV_CMD_DO_SET_MODE">
                    <description>Set system mode.</description>
                    <param index="1">Mode, as defined by ENUM MAV_MODE</param>
                    <param index="2">Custom mode - this is system specific, please refer to the individual autopilot specifications for details.</param>
                    <param index="3">Empty</param>
                    <param index="4">Empty</param>
                    <param index="5">Empty</param>
                    <param index="6">Empty</param>
                    <param index="7">Empty</param>
               </entry>

Both looks mavlink compliant. (I continue my investigation).

KiwiHC16, 2015-07-24:
grep -r MAV_CMD_DO_LAND_START *
ArduPlane/release-notes.txt: - added support for MAV_CMD_DO_LAND_START
ArduPlane/GCS_Mavlink.cpp:        case MAV_CMD_DO_LAND_START:
ArduPlane/commands_logic.cpp:    case MAV_CMD_DO_LAND_START:
ArduPlane/commands_logic.cpp:    case MAV_CMD_DO_LAND_START:
libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/ardupilotmega.h: ...
libraries/AP_Mission/AP_Mission.cpp:    case MAV_CMD_DO_LAND_START:   
libraries/AP_Mission/AP_Mission.cpp:    case MAV_CMD_DO_LAND_START:   
libraries/AP_Mission/AP_Mission.cpp:        if (tmp.id == MAV_CMD_DO_LAND_START) {

and with grep -r MAV_CMD_DO_SET_MODE *
AntennaTracker/GCS_Mavlink.cpp:            case MAV_CMD_DO_SET_MODE:
APMrover2/GCS_Mavlink.cpp:        case MAV_CMD_DO_SET_MODE:
ArduPlane/GCS_Mavlink.cpp:        case MAV_CMD_DO_SET_MODE:
libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/ardupilotmega.h: ...	libraries/GCS_MAVLink/message_definitions/common.xml: ...

So my conclusion is:
'land' command in MAVProxy is associated to  MAV_CMD_DO_LAND_START which is implemented only in Plane code.
To land with copter/plane/... then we can use the MAV_CMD_DO_SET_MODE in common code.
It works for copter. Can someone verify that it works with plane ?