ArduPilot/MAVProxy #169: Getting MAVProxy error when running "sim_vehicle -w"

gmorph, 2015-07-26:
ERROR in command: 'MPState' object has no attribute 'logqueue'
Traceback (most recent call last):
  File "/home/grant/.local/lib/python2.7/site-packages/MAVProxy-1.4.24-py2.7.egg/EGG-INFO/scripts/mavproxy.py", line 473, in process_stdin
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_param.py", line 246, in cmd_param
    self.pstate.handle_command(self.master, self.mpstate, args)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_param.py", line 207, in handle_command
    self.mav_param.load(args[1], param_wildcard, master, check=False)
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/mavparm.py", line 90, in load
    self.mavset(mav, a[0], a[1])
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/mavparm.py", line 22, in mavset
    mav.param_set_send(name.upper(), float(value))
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/mavutil.py", line 405, in param_set_send
    parm_name, parm_value, parm_type)
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9661, in param_set_send
    return self.send(self.param_set_encode(target_system, target_component, param_id, param_value, param_type))
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 7547, in send
    self.send_callback(mavmsg, *self.send_callback_args, **self.send_callback_kwargs)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_link.py", line 188, in master_send_callback
    if mtype != 'BAD_DATA' and self.mpstate.logqueue:
AttributeError: 'MPState' object has no attribute 'logqueue'
ERROR in command: 'MPState' object has no attribute 'logqueue'
Traceback (most recent call last):
  File "/home/grant/.local/lib/python2.7/site-packages/MAVProxy-1.4.24-py2.7.egg/EGG-INFO/scripts/mavproxy.py", line 473, in process_stdin
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_param.py", line 246, in cmd_param
    self.pstate.handle_command(self.master, self.mpstate, args)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_param.py", line 127, in handle_command
    master.param_fetch_all()
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/mavutil.py", line 383, in param_fetch_all
    self.mav.param_request_list_send(self.target_system, self.target_component)
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9585, in param_request_list_send
    return self.send(self.param_request_list_encode(target_system, target_component))
  File "/home/grant/.local/lib/python2.7/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 7547, in send
    self.send_callback(mavmsg, *self.send_callback_args, **self.send_callback_kwargs)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_link.py", line 188, in master_send_callback
    if mtype != 'BAD_DATA' and self.mpstate.logqueue:
AttributeError: 'MPState' object has no attribute 'logqueue'
Log Directory: 
Telemetry log: mav.tlog

peterbarker, 2015-07-28:
This issue should have been fixed with:

https://github.com/Dronecode/MAVProxy/commit/c9729f59474c9d4a144b25186740f6c7a6ce88b2

Perhaps there hasn't been a release since then?

gmorph, 2015-07-28:
Confirmed its fixed in the diydrones repository of MAVProxy.  My git tree was still pulling from Tridge's original.
Thanks, Grant.

closed 2015-07-28 by gmorph