ArduPilot/MAVProxy #172: Add a note on how to set the baud rate from running MAVProxy

hamishwillee, 2015-08-03:
We document how to specify the baudrate for a serial connection when starting MAVProxy but not when it is running - which is a problem on windows when you're starting mavproxy from double-click.

It would be good to add a note on how to do this, minimally in http://tridge.github.io/MAVProxy/link.html

Tridge suggested that calling ``set baudrate XXXXX`` before using link add is probably the easiest way. It would also be useful to show the syntax for adding with link add, and also make it clear that the default baud rate is 57600

stephendade, 2015-08-15:
Fixed in #180

stephendade, 2015-08-15:
Also, the documentation is actually at http://dronecode.github.io/MAVProxy/  :)

hamishwillee, 2015-08-17:
Thanks Stephen

1. Change looks good to me but is not yet live
2. Tridge version is still live, and is what gets hit when I do a search on google - perhaps get him to remove it?

stephendade, 2015-08-17:
Changes live now.

closed 2015-10-05 by stephendade