ArduPilot/MAVProxy #179: MAVProxy not respecting the windows path inside quote marks

hamishwillee, 2015-08-14:
This was posted as a [DroneKit issue](https://github.com/dronekit/dronekit-python/issues/291), but I think is probably an issue with MAVProxy.

The following start command fails:

```
api start  “C:\Program Files (x86)\MAVProxy\examples\vehicle_state\vehicle_state.py"
```
It returns the following error:
```
STABILIZE> api start "C:\Program Files (x86)\MAVProxy\examples\vehicle_state\vehicle_state.py"

STABILIZE> Exception in APIThread-0: [Errno 2] No such file or directory: '"C:\\Program'

Traceback (most recent call last):

  File "C:\PROGRA~2\MAVProxy\droneapi\module\api.py", line 333, in run

STABILIZE>    self.fn()

  File "C:\PROGRA~2\MAVProxy\droneapi\module\api.py", line 625, in <lambda>

    APIThread(self, lambda: execfile(args[1], g), args[1])

IOError: [Errno 2] No such file or directory: '"C:\\Program'
```

Mavproxy or dronekit does not interpret the directory  correctly. Tried everything I could with other path syntaxes without success.

This originates here: https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/128#issuecomment-131135888 

I have verified issue with 1.5 Windows Installer and with a normal Python installation from PIP. NOt sure which MAVProxy windows installer, but would be fairly recent.

hamishwillee, 2015-08-14:
@stephendade Help me oh mighty guru!

mrpollo, 2015-08-14:
Identified the problem here https://github.com/Dronecode/MAVProxy/blob/master/MAVProxy/mavproxy.py#L444

```
split()
```

The current implementation of reading STDIN splits the arguments by whitespace (default split behaviour), not sure if this is a bug or intended, but it does affect DroneKit since we somehow depend on paths to files

stephendade, 2015-08-16:
Fixed in #182

hamishwillee, 2015-08-16:
Thanks @stephendade. When do you think there might be a new [Windows Installer](http://firmware.diydrones.com/Tools/MAVProxy/) we can try this out on?

stephendade, 2015-08-17:
Whenever Tridge gets around to merging the pull request and up-versioning MAVProxy :)
I'll talk to him and see if we can get it done soon.

stephendade, 2015-09-30:
And this has been merged into MAVProxy 1.3.32

closed 2015-09-30 by stephendade

mrpollo, 2015-10-07:
thanks @stephendade

hamishwillee, 2015-10-07:
From me too @stephendade !