ArduPilot/MAVProxy #181: Error importing modules in scripts called within MAVProxy

HuguesD, 2015-08-15:
Hi,
I try to run a simple script from MAVProxy (via command:  api start script.py) which contains this code:
###################################
import sys, pygame, pygame.mixer

pygame.init()

size = width, height = 600, 400
black = 0,0,0

screen = pygame.display.set_mode(size)
tux = pygame.image.load("logo.bmp")

x = 0
y = 0
r = 0
g = 0
b = 0
while 1:
	for event in pygame.event.get():
		if event.type == pygame.QUIT:
			sys.exit()
		elif event.type == KEYDOWN:
			sys.exit()
	screen.fill((r,g,b))	
	screen.blit(tux,(200,200))
	screen.blit(tux,(x,y))
	pygame.display.flip()
	x = x+1
	y = y+1
	if r == 255:
		r1 = -1
	elif r == 0:
		r1 = 1
	r = r+r1
##############################
It stop on an Import Error : no module named mixer
Although the sys.path is correctly set and although the file pygame.mixer.pyd is in the script directory
I'm under windows.
Any idea how to solve this issue ?

stephendade, 2015-08-16:
This is a dronekit-python (https://github.com/dronekit/dronekit-python/) issue, so you would more support there.

But in answer to your question, if you used the Windows installer it does not use your python environment - so any libraries in your python environment will not be accessible by MAVProxy.

You would need to put the pygame libraries in the MAVProxy folder for it to work.

HuguesD, 2015-08-16:
Thx Stephen. This is what I tried : i installed separately pygame with its own installer. I copied then all of the libraries of pygame to the MavProxy directory. But still getting this module not found error. It is like Mavproxy sees only the modules that are integrated with it. 
I wish there was a simple way to get all of these dependencies/directories right.
For now, I'm dropping pygame and will use Wx instead, as Mavproxy seems to know these libraries...

stephendade, 2015-08-16:
hmm ... could you try copying "C:\Python27\Lib\site-packages\pygame" folder to the MAVProxy folder? That should work.

HuguesD, 2015-08-16:
What I did now that works as I want it to:
-I uninstalled everything (MavProxy, Pygame, Wynpython) because I wanted to start from a clean base
-I installed Winpython (C:\Program Files (x86)\WinPython-64bit-2.7.6.4)
-I installed droneapi , which inludes MavProxy. (C:\Program Files (x86)\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\Lib\site-packages\MAVProxy)
-I installed WxPython (C:\Program Files (x86)\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\Lib\site-packages\wx-3.0-msw\wx)
-I downloaded from Github a zip containing the droneapi examples (.py scripts) and placed the examples folder in the MavProxy directory

Since folders & path are not recognized, I did a windows batch file that launches for me automatically Mavproxy & loads dronapi, giving me the command window with the MavProxy prompt (executing in the MavProxy directory). I then just need to make sure my scripts are placed in the MavProxy directory and then launch "api start script.py".
Here is the content of the batch file (named : MavHug.bat) that I use and works for me (my vehicle being on com10):
--------
@echo off
set WINPYDIR="C:\Program Files (x86)\WinPython-64bit-2.7.6.4\python-2.7.6.amd64"
set MAVPROXYDIR="C:\Program Files (x86)\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\Lib\site-packages\MAVProxy"
set PATH=WINPYDIR; MAVPROXYDIR; %WINPYDIR%\Lib\site-packages\PyQt4;%WINPYDIR%\;%WINPYDIR%\DLLs;%WINPYDIR%\Scripts;%WINPYDIR%\tools;%WINPYDIR%\tools\gnuwin32\bin;%WINPYDIR%\tools\mingw32\bin;%PATH%;%WINPYDIR%\tools\TortoiseHg

start cmd.exe /k "%WINPYDIR%\python.exe %MAVPROXYDIR%\mavproxy.py --master=com10"

GOTO End1

IF %1.==. GOTO No1

:No1
  ECHO Usage : mavHug 'com port' 
  ECHO Example : mavHug com10 
  ECHO Missing param 1 : parameter: the com port your mavlink vehicle is connected to (ex : com10)
  pause
GOTO End1

:End1

HuguesD, 2015-08-16:
forgot to mention I also created a mavinitscr file in my windows user home directory/appdata/local , containing the line : module load droneapi.module.api

stephendade, 2015-08-17:
MAVProxy does include some parts of pygame for the joystick/rc controller handling.
The rationale with the windows exe is to create a simple click-to-install program for users (avoiding the complexities with python installs and various dependencies). It works quite well if you're not developing new modules/scripts that require external libraries.
Given the development that people are doing with droneapi, I'll need to take a look at if there's a good way to implement external library dependencies without messing up the install.

HuguesD, 2015-08-17:
Yes, thank you for your support. As Randy says, undwer windows it is still a bit green but I'm sure it will be soon much better with the gathered experience.

stephendade, 2015-08-19:
On the Gitter chat, a user came up with this solution:
> ok, I solved the issue when importing your own libraries, I had to add some code to my scripts...
```
import os, sys
sys.path.append(os.getcwd())
import yourownmodule
```
The next thing to solve is using threads inside mavproxy...

Does this work for you?

HuguesD, 2015-08-19:
Well, I confirm that I tried the following with success:
-Deinstalled everything (clean sheet to start with is especially important in Windows...)
-Reinstalled in order : Python 2.7, MavProxy, DroneApi, Wx
-Wx can be imported without problem and I now succeed to ùake my own GUI on top of MavProxy, that's fine
-So I decided to goive another chance to pygame. And I think I foudn what the problem was. On the pygame web download page, there is no compiled install package for the combination of python 2.7 and 64bits OS. So they redirect you to a university web site where a WHL file is made available for python 2.7 and 64 bits Win OS. I did not know what to do with a WHL file until i googled for a while and found out I just needed to open a python command window under windows and type "pip install pygamexxx.WHL". That did the trick and installed pygame in such a way I can now import pygame without issues. It is weird because in my previous manual attempts to install pygame, I copied the libs everywhere to avoid path problems and that did not work. So the pip install does something more to make it work but I do not know what exactly.

HuguesD, 2015-08-19:
I guess it all not works perfectly after all. I get a systematic "timeout setting" a parameter to a value with the documented command : vehicle.parameters['PARAM']=value (and then vehicle.flush())
Even with the examples scripts that come with MAVProxy, all works ok except this timeout when setting a parameters. I use a real pixhawk board connected by USB cable on com10.
ANy idea what could be the explanation for this timeout problem ?

amilcarlucas, 2018-10-17:
That parameter timeout was a bug that got fixed some time ago

amilcarlucas, 2018-10-17:
@tridge please give me permissions to close issues

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas