ArduPilot/MAVProxy #183: Add search support to the docs?

hamishwillee, 2015-08-17:
Hi @stephendade 

Might be worth moving MAVProxy to use an SSG that supports search and ideally sidebar navigation. The flat site structure is not particularly useful for navigation and it is much easier to work with markdown or reStructured text than html.

I recommend Sphinx or Jekyll. Happy to do the migration work if you want this.

Regards
H

stephendade, 2015-08-17:
Sounds like a good idea!
I'm not particularly well versed in setting up Sphinx or Jekyll environments, so I'll take you up on your offer of help.

hamishwillee, 2015-08-18:
Great. Do you have a strong preference for mardown or reStructured text for your inputs? If not, I recommend Sphinx, which uses restructured text, and probably the read the docs theme so it will looks something like (http://nailgun.readthedocs.org/en/latest/)

If you much prefer markdown then we'd probably look at Jekyll - maybe even this default theme http://jekyllrb.com/docs/home/

stephendade, 2015-08-18:
Sphinx looks good!

hamishwillee, 2015-08-19:
@stephendade I'm on it. Do you need the url filenames to be unbroken? I can keep them the same, but personally I think it would be nicer to add a directory structure, which will be definition mean that the urls change. Thoughts?

hamishwillee, 2015-08-19:
Hi @stephendade 
Checkout this first [build on my dropbox](https://dl.dropboxusercontent.com/u/3067678/share2/mavproxy/index.html). I split out the "Downloading" and installation from the front page into its own topic and put the license text in its own page linked from sidebar too.

I also put the text you used to have on every page in the front page "acknowledgements" section "MAVProxy is maintained by Andrew Tridgell and documentation is by Stephen Dade"

This is the "out of the box" read the docs theme. If you want some modifications to look and feel all that can be done "easily" is hiding that link to read the docs theme at the bottom and you can also make the sidebar "sticky" (though I don't plan to)/.If you want to modify this more significantly that is possible - but we'd need to modify the theme itself. Easy enough to do, but you'll need to provide direction.

Have a play, let me know what you think. I will then submit the sources for you to play with yourself and make live.

Cheerio
H

stephendade, 2015-08-19:
That looks really good @hamishwillee! A directory structure makes sense, as I expect the documentation to grow over time as more of MAVProxy is documented (and more modules created).

Also - I've fixed up the old documentation site at http://tridge.github.io/MAVProxy to redirect to the new site. So we only need to keep the documentation updated here.

hamishwillee, 2015-08-20:
Hi @stephendade 

OK, so I have submitted pull request #184. This contains the sources (now inside directories) and a vagrant file that you can use to build them. Hope this all OK!

See below for instructions on building. Apologies if you know all about vagrant, I'm writing this as if you don't!

The steps to building are:

* Download [Vagrant](http://www.vagrantup.com/downloads.html). I tested on Windows.
* Clone the repo and navigate to /docs/ directory
* In command prompt in /docs directory enter ``vagrant up``. This will fetch everything you need to build sphinx, and then build the docs. It will take a long time first time as it has to fetch a virtual machine image
 
 This will build the source in /docs/build/html. You can copy these files to the gh-pages branch to publish. I haven't done that for you because I want to make sure we've resolved any build issues.

The beauty of this system is that the tools for building the sources (Sphinx) are in a virtual machine. The tools act on the folders in the docs/directory which is shared with windows. This means that you can edit the files in windows, then build them using vagrant without having to think about the toolchain much at all.

In reality you will probably need to build multiple times. In this case you SSH into the vagrant instance. Assuming you already have SSH on your system you can do this (and then build) using:

```
vagrant ssh
cd /vagrant 
make clean      #Optional. If you want to do a full rebuild
make html
```
If you come back to the same folder and want to build at another time, you can restart the process using ``vagrant up``. It will be much faster second time as the image needed will be present and provisioned.

You should also check out the [reStructured Text primer](http://sphinx-doc.org/rest.html)

Let me know how this goes for you ... I'd love to see the new docs up!

stephendade, 2015-08-20:
I ended up using my linux box for compilation. Nice and easy :)
Documents look good - but there's a few issues I've mentioned in the pull request that need to be fixed.

hamishwillee, 2015-08-20:
I've answered that question (I think). Drop me an email or skype if you want to chat about it or if my response doesn't make sense.

closed 2015-09-30 by stephendade