ArduPilot/MAVProxy #200: unable to run installed version.

AndKe, 2015-09-29:
after install 
andre@Loke:~/prog$ python setup.py build install --user

I am unable to run it:
- the mavlink.py works from the source folder.

```
andre@Loke:~/prog$ mavproxy.py 
Auto-detected serial ports are:
/dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00
Connecting to /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00
Failed to connect to /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00 : 
Logging to mav.tlog
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 923, in <module>
    load_module(m, quiet=True)
  File "/usr/local/bin/mavproxy.py", line 259, in load_module
    module = m.init(mpstate)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_rally.py", line 325, in init
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_rally.py", line 17, in __init__
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/mp_settings.py", line 108, in __getattr__
AttributeError
andre@Loke:~/prog$ 
```

stephendade, 2015-09-30:
Have you upgraded to the latest version of mavlink?
<code>pip install pymavlink -U</code>

AndKe, 2015-10-02:
Thank you.
After some fooling around, uninstalling pip installed mavproxy, installing locally build one, things work again.  this was clearly my fault.

closed 2015-10-02 by AndKe