ArduPilot/MAVProxy #201: mavlogdump.py : unable to export .bin to .csv

AndKe, 2015-09-29:
this works /(see types)
mavlogdump.py "/home/andre/test2, trimdata.BIN" --types POS,CAM 

this does not:
mavlogdump.py "/home/andre/Linux/Observer test2, trimdata.BIN" --types POS,CAM --format csv

once format = csv, I get an error with first type mentioned, regardless of the actual type(s)

Traceback (most recent call last):
  File "/usr/local/bin/mavlogdump.py", line 73, in <module>
    fields += [type + '.' + x for x in inspect.getargspec(getattr(mavutil.mavlink, typeClass).__init__).args[1:]]
AttributeError: 'module' object has no attribute 'MAVLink_pos_message'

stephendade, 2015-09-30:
mavlogdump is part of mavlink (https://github.com/mavlink/mavlink), so you will have better luck posting over there.

peterbarker, 2017-12-16:
I believe this is fixed now, anyway!

closed 2017-12-16 by peterbarker