ArduPilot/MAVProxy #202: Docs - a few issues around setup/installation

hamishwillee, 2015-09-30:
Just noted the following issues with docs

1. [Download & Install](http://dronecode.github.io/MAVProxy/html/getting_started/download_and_installation.html) 
    * [(Windows)](http://dronecode.github.io/MAVProxy/html/getting_started/download_and_installation.html#windows)
      * ``Please note that the “missionedit” ...`` - This should be a warning and should link to the issue number for tracking - do you have one?
      * Section should explain that since you're not starting mavproxy with connection you need to use ``link add`` in the MAVProxy command prompt to add the connection
      * Section should probably note how you add modules to this installation
      * Section should link to the "development" guide for people who want to work off a python command prompt.
    * [(Linux)](http://dronecode.github.io/MAVProxy/html/getting_started/download_and_installation.html#linux) 
      * has some rubbish at end of dependencies: *python-pip``*
      * For DroneKit we also needed "python-dev python-serial python-pyparsing". I suspect at least the first one is actually a MAVProxy dependency?
    * [(mac)](http://dronecode.github.io/MAVProxy/html/getting_started/download_and_installation.html#mac)
      * This states ``OSX should require the same prerequisites as for the Windows installations.`` but the Windows section has an installer - does not list the pre-reqs. I think you should replace this with the [docs in DroneKit here](http://python.dronekit.io/guide/getting_started.html#installing-dronekit-on-mac-osx) (minus the DK itself). Shift it round so that the homebrew stuff is at the end.

1. [Development Environment](http://dronecode.github.io/MAVProxy/html/development/mavdevenv.html)

   * Linux - the download source is old tridge repo ``git clone https://github.com/tridge/MAVProxy.git`` - should do a search for tridge on whole repo.
   * Windows - is pygame really needed, and do Numpy/Pillow/OpenCV really need to be done first? FYI in DK we started from a more complete distribution of Python to reduce the dependencies - but this is better :-)
   * It might be worth marking up file paths - e.g. (C:\Python27\Scripts) as :file:\`C:\Python27\Scripts` or in bold (e.g. **C:\Python27\Scripts**)

hamishwillee, 2015-09-30:
Also it was recently reported that libxml2-dev and libxslt-dev are MAVProxy dependencies on RPi/Raspian. Is this so? Buzz indicated maybe this was added with  mavexplorer?

hamishwillee, 2015-09-30:
Front page is missing the screenshots for Linux and Windows

http://dronecode.github.io/MAVProxy/html/index.html

stephendade, 2015-09-30:
Yep, libxml2 is a prereq for MAVExplorer. Can be installed via 'pip install lxml'. I'm not aware of libxslt being necessary though.

Screenshots I'm intending to update over the weekend with some fresh ones showing off the updated MAVProxy.

Thanks for the other comments - I'll add them in soon.
-Numpy/Pillow/OpenCV need to be installed first, as some of the 'pip install X' libraries require them as prerequisites
-Pygame is only required for the joystick module.

hamishwillee, 2015-09-30:
Thanks very much.

> -Numpy/Pillow/OpenCV need to be installed first, as some of the 'pip install X' libraries require them as prerequisites

In DroneKit we didn't install Pillow at all. Numpy was part of our WinPython install so by default done first. OpenCV was done after and that worked fine. 

What you've done by listing the dependencies is great because it means that people can use whatever Python distribution they like. Our instructions are a bit easier though, because they reduce the scope by choosing a single distro to work with. Just nothing - not suggesting you change anything.

Let me know if you need help with any of this.

stephendade, 2015-10-05:
And done in #204!

closed 2015-10-05 by stephendade

hamishwillee, 2015-10-05:
@stephendade - excellent. I didn't check "everything" but I believe most of my concerns were met! I did see a few new issues #205 , #206

PR #204 (2015-10-05): Big doco fix
Lots of new updates:
-Fixed #202 
-Better layout of modules section
-Added arming/disarming