ArduPilot/MAVProxy #209: Running with SITL simulation

petoalbert, 2015-10-09:
Running `mavproxy.py --master localhost:14556` to connect to the 14556 mavlink port of PX4 in SITL does not work, since the code in mavproxy_link.py opens the connection as a socket that binds to the specified port instead of connecting to it. For now I have changed *mavproxy_link.py* to invoke `mavutil.mavutil_connection(...,input=False)` from `link_add(self,device)`, and it works now, but I know that it's not the right way to solve this problem. I think a new command line option could be added to specify whether the master udp connection should be opened as a listening port or not. I would like to know what you think of it.

(I know about the `--sitl` option, but looking in the sources, its effect seems to be related only to rc override packets.)

stephendade, 2015-10-10:
Is this for the APM or PX4 flight software?
MAVProxy definitely supports APM, but it's not been rigorously tested on PX4.

petoalbert, 2015-10-10:
This is for the PX4.

stephendade, 2015-10-10:
Ok, I'll put this on the "todo list". May take a few weeks for me to get around to doing.

stephendade, 2015-12-02:
I've looked into this. Using `mayproxy.py localhost:14560` `mayproxy.py localhost:14556` works fine for me. This is using the latest source for both MAVProxy, mavlink and PX4.

Are you still having issues?

closed 2016-05-24 by stephendade