ArduPilot/MAVProxy #211: Image Processing with cuav - Image source

Georacer, 2015-10-11:
URL: ../cuav_module/index
Title: Image Processing with cuav

This isn't an urgent request or fix - I'd just like to see this documentation as complete as possible. It's a really nice tool.

So, I'd like to see more documentation on the image source for the cuav. Is it a folder? Is it a ttyUSB* camera?

Thanks.

stephendade, 2015-10-12:
The image source is a USB camera for the "live" version. There's also an offline version of cuav that works with folders and flight logs.
I'll have to look carefully through the code to find out which cameras the online version works with and the specific file/folder structure required for the offline version.

stephendade, 2015-10-14:
Issue partly fixed now in #213 . I've put in information for camera compatibility. Still need to look at the folder source.

Georacer, 2015-10-14:
Nice. This clearly communicates the hardware requirements. Thanks!

Georacer, 2015-10-14:
If cuav will not support pulling images from a folder any time soon, I consider this issue closed. Shall I?

stephendade, 2015-10-14:
Correct. Currently you only only use folders when running cuav offline.

It's on the long-term todo list to explain how to do that.

closed 2015-10-14 by Georacer