ArduPilot/MAVProxy #215: Simulating multiple vehicles

iivanoo, 2015-10-19:
How can we simulate multiple vehicles communicating via MAVLink messages?

stephendade, 2015-10-20:
Take a look at http://dev.ardupilot.com/wiki/setting-up-sitl-on-linux/ and http://dronecode.github.io/MAVProxy/html/analysis_and_simulation/simulation.html for simulation of Vehicles.

At this point in time, MAVProxy only supports 1 vehicle per instance.

iivanoo, 2015-10-20:
Thanks Stephen, I will find a solution for having a shared world representation among different instances of MAVProxy. Do you have any initial pointers to suggest? I already studied and used SITL, MAVProxy, ROS.

monkeypants, 2015-10-20:
What about Flightgear? I've "flown" a chaser plane to watch my sitl sim in Flightgear, don't see why you couldn't have an arbitrary number of sitl sims in the same Flightgear world.

iivanoo, 2015-10-20:
Thanks, I will start my experiments from Flightgear+SITL.

monkeypants, 2015-10-20:
Are you sure all of mavproxy supports only one vehicle per instance @stephendade? E.g. GCS should support multiples (right?). Is it the case perhaps that only one SITL module per instance, but multiple instances could stream through pipes or socket links into a central instance? Maybe the Flightgear world is redundant. But fun, sim is more fun with "synthetic vision" ;)

stephendade, 2015-10-20:
Definitely 1 vehicle per instance (though there can be multiple connections to that vehicle). It's something that @tridge is thinking of fixing up for the 2016 UAV Challenge if we decide to use a relay aircraft.
There's nothing stopping multiple instances of MAVProxy being used, but there would be no data sharing between them.

stephendade, 2015-10-20:
To handle multiple vehicles, there would have to be some major changes to the MAVProxy source. Hence it may take some time for this feature to appear.

iivanoo, 2015-10-20:
Thanks guys for the clarifications, I will go with multiple instances of mavproxy with a "shared world".
Just for the visualization part, I will proceed by streaming positioning data from the mavproxy instances to Google Earth, so that I can see all the vehicles together.

ccnavio, 2017-06-13:
@iivanoo Did you ever get this working?

iivanoo, 2017-06-14:
Yes, with the help of one of the PhD students we managed to have 3 drones simulated + Google Earth. A demo is available here: http://www.flyaq.it/index.html#demo

gmorph, 2017-06-15:
Did you have to make code updates to support this and if so are they somewhere in github.  Would love to be able to bring these across into the main project.

Thanks, Grant.

iivanoo, 2017-06-15:
Hi, as far as I remember we did not need to extend the code of MAVProxy.

Alex-Horlock, 2019-08-21:
Hi, I've been attempting to attach multiple simulated vehicles (using the arduplane.exe built from Ardupilot source code) via tcp connections to MavProxy 
(e.g. mavproxy.exe --master=tcp:127.0.0.1:5770 --master=tcp:127.0.0.1:5780 --master=tcp:127.0.0.1:5790 --out=udpout:127.0.0.1:14550) 

Does anyone know if this is possible? All the state data comes through nicely to an instance of QGC connected to the UDP out but when trying to task the platforms data only goes to the first tcp connection in the list. 

To setup the 3 sims I am using:
arduplane.exe --model plane -I 'N' --defaults planeN.parm
(where N is the instance number and planeN.parm contains a SYSID_THISMAV of N.)

Any help would be greatly appreciated. Thank you.