ArduPilot/MAVProxy #226: Exits if you start with a --master=/dev/ttyXXX that doesnt exist

mikerob, 2015-11-21:
If you start with a master tty that doesn't exist, Mavproxy crashes out, but if you add a link that doesn't exist during runtime, it handles it fine.

This is particularly problematic for mavgateway - it means you can't just set "all" of the possible masters and go with whatever you've got connected, you HAVE to list the right one, which requires script editing.

peterbarker, 2016-09-21:
This is a rough implementation of a "force-connect" option: https://github.com/ArduPilot/pymavlink/compare/master...peterbarker:force-connect?expand=1 with an associated MAVProxy commit here: https://github.com/ArduPilot/MAVProxy/compare/master...peterbarker:force-connect?expand=1

I'm not sure we shouldn't actually infer force-connect from autoreconnect; you will get a "device is dead" warning instantly in that case.

Your thoughts?

I haven't seen mavgateway, but mavlink_hub.py (
https://github.com/dronekit/dronekit-python/blob/source-system-filtering/examples/avoidance/mavlink_hub.py) does without it.

mikerob, 2016-09-21:
It looks like mavlink_hub is a slightly different use case - it's not using MAVproxy at all.  For my application I need BOTH a bespoke python application and MAVproxy since I want to split the stream and have my python application process it as well as being able to connect via a standard GCS.  This was the motivation for another request to provide a stream output function within dronekit python - so that you could run both an application and a GCS connection.

Specifically, I have a 900MHz feed coming into a Raspberry Pi running mavgateway (MAVproxy in daemon mode) which displays GPS location and RSSI on an I2C LCD, and acts as a wifi host to enable you to also connect with a PC GCS. I run MAVProxy connected to the serial 900MHz, and output two UDP streams, one of which is picked up by the python app running the LCD, and the other can be grabbed by the PC GCS over wifi.  My problem was that I didn't start the Pixhawk and telemetry radios before the Raspberry Pi booted up, it would fail to connect and crash. 

Regarding the specific implementation, it sounds like a good start if it will keep trying to reconnect rather than failing out.  I see there's a "retries" in there, but I can't tell how that's implemented. If there were a "retries" parameter, I'd like a "-1" option that keeps trying forever.

It's been a while since I was using this, so I've forgotten precisely how to use MAVProxy, and the MAVProxy documentation is gone - http://dronecode.github.io/MAVProxy/ and tridge.github.io/MAVProxy both go nowhere,

It would be good to be able to list multiple "master" links and have the system either: 
- Keep iterating through the list until it finds one that works, and/or
- Run through the list connecting to ALL of the ones that work

The project that used this will be kicking off again soon, so I should be able to do some proper testing and thinking in a few weeks time.