ArduPilot/MAVProxy #227: Running sim_vehicle.sh is running the mavproxy.py script which is throwing a lot errors.

rahulraghu94, 2015-11-23:
These included a few errors in print statements that were in python2 format which i changed with not much problem.
The issue i am facing now is in importing the module Queue from line no.12 and can't seem to find a work around.

stephendade, 2015-11-23:
Which Python version are you running? MAVProxy only works with Python2 at this time.

rahulraghu94, 2015-11-23:
I'm running both versions side by side. I used to use python3 for openCV related projects.
I changed the header of mavproxy.py to have #/bin/pythin2. I used pip to install mavproxy and pymavlink and its installed in the python3 directory and is not being detected. Post fixing that, it shows errors for importing Queue. I tried a pip install Queue to find that it returns that Queue does not exist

rahulraghu94, 2015-11-23:
pip install queue                                                                                                                                         *[next] 
Collecting queue
  Could not find a version that satisfies the requirement queue (from versions: )
No matching distribution found for queue

stephendade, 2015-11-24:
The Queue module is part of the base install of Python, so this error would indicate a problem with your Python installation.
I've just tried with the latest source of MAVProxy and Ardupilot and I have no issues running it.

monkeypants, 2015-11-24:
@rahulraghu94, can you reproduce this problem with a fresh Python 2 virtualenv (created with --no-site-packages)?

peterbarker, 2016-09-19:
@rahulraghu94 Please reopen this issue if you can reproduce this problem as @monkeypants suggests.

closed 2016-09-19 by peterbarker

aviogit, 2017-12-05:
It happened the same to me just now. For whatever reason (I'm new to python so I'm still not fully understanding), running `sim_vehicle.py -w` from [SITL](http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#setting-up-sitl-on-linux) was launching mavproxy.py using Python3. I was starting to fix bugs such like the Queue one with `import queue as Queue` when I figured out that mavproxy.py was written for Python2. Now I'm trying to understand why Python3 is called at all.

rahulraghu94, 2017-12-05:
@aviogit , you'll have to create a virtual environment specifically for python2 and then try to use Mavproxy.

red [this](http://docs.python-guide.org/en/latest/dev/virtualenvs/) as a guide. It should help :)
Good luck!

aviogit, 2017-12-05:
I solved the problem with a dirty hack...

In `ardupilot/Tools/autotest/sim_vehicle.py` I just prefixed the `mavproxy.py` launch with the correct Python2 interpreter:

>         cmd.append("python2")
>         cmd.append("/usr/local/bin/mavproxy.py")
>         #cmd.append("mavproxy.py")

aviogit, 2017-12-05:
Of course you are right, but I'm not going to ask for a pull request of my dirty hack :)
And maybe it can help some poor soul who stumbles upon this error like I did. Anyway I'll keep in mind that an "official way" of handling Python2/3 conflicts exists.

rahulraghu94, 2017-12-05:
Maybe you could use your experience to edit the manual/readme file and create PR of that instead. That would actually be a better contribution in my opinion, given that you've experienced this problem. You can help people this way and still maintain the official stuff.

Just my $0.02! Good luck!

peterbarker, 2017-12-05:
On Tue, 5 Dec 2017, Rahul Raghunath wrote:

> Maybe you could use your experience to edit the manual/readme file and
> create PR of that instead. That would actually be a better contribution in
> my opinion, given that you've experienced this problem. You can help people
> this way and still maintain the official stuff.
> 
> Just my $0.02! Good luck!

Rumour has it that MAVProxy did at some stage work with python3, but we've 
lost that somewhere along the line.

There's really two things we need...
- patches to make it work with Python3 again
- some sort of simple CI system to make sure it basically works so we 
don't break again

rahulraghu94, 2017-12-06:
I think writing a CI system for python2 first would be the way to go. You can then port the tests to python3 which *should* weed out all the compatibility problems!

stevegloor, 2017-12-23:
I got the same error that I fixed with import queue as Queue.  I created a new 2.7 virtual environment and the error still occurred.  However did not create one without site packages.  This was on a clean install of Ubuntu 16.04LTS.

I also had done a full install of OpenCV on a python3 virtual environment so I am not sure if that is stuffing it up.