ArduPilot/MAVProxy #236: No message in console when calibrating accelerometer

jberaud, 2015-11-27:
Using Mavproxy or apm planner, no message is displayed when calibrating accelerometer

git bisect pointed to https://github.com/diydrones/ardupilot/pull/2952 as the responsible

billbonney, 2015-11-27:
you will need a version of APM Planner 2.0 built from this branch for running master ardupilot https://github.com/diydrones/apm_planner/tree/statustext-wip with the SEVERITY WARNINGs updated. (it's a work-in-progress as it doesn't detect the APM version yet so it know when to apply the fix)

jberaud, 2015-11-30:
Thanks, I will take a look at the modifciations. What about mavproxy?

jberaud, 2015-11-30:
Tridge said that this was working fine with px4, so it is either a linux or a bebop issue.
Real time seems to have to do with it since when stepping into the code with gdb, the messages are printed correctly.

jberaud, 2015-12-11:
Ok, found it.
It is due to the limit of 300 bytes in udp recv in pymavlink. It has already been fixed on master but the version installed by pip didn't have it.
Peter's pr revealed the issued because it is sending lots of packets rapidly.

closed 2015-12-11 by jberaud