ArduPilot/MAVProxy #237: Mavproxy :-Exception in thread main loop , An existing connection was forcibly closed by remote host on windows

sakshamgupta006, 2015-12-14:
I am getting the following error in Mavproxy after type in the command sim_vehicle.sh -j4 --map in Arduplane directory

Connect tcp:127.0.0.1:5760 source_system=255
Loaded module map
Running script C:\Users\Saksham\AppData\Local\MAVProxy\mavinit.scr
-> set moddebug 2
-> module load graph
Loaded module graph
-> module load console
Loaded module console
Loaded module help
Log Directory:
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Exception in thread main_loop:
Traceback (most recent call last):
File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\threading", line 810, in __bootstrap_inner
File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\threading", line 763, in run
File "", line 717, in main_loop
File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 374, in wait_heartbeat
File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 339, in recv_match
File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 302, in recv_msg
File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 885, in recv
error: [Errno 10054] An existing connection was forcibly closed by the remote host

stephendade, 2015-12-14:
I've just tried using the latest MAVProxy and APM and it works fine for me.

Are you using the latest MAVProxy (http://firmware.diydrones.com/Tools/MAVProxy/) and APM source code?

How are you using sim_vehicle.sh - is it via Cygwin on a Windows box?

sakshamgupta006, 2015-12-14:
i am using cygwin for the commands
but I have initially installed some previous version of Mavproxy but then I re-installed it with the latest version of Mavproxy over the previous one..

stephendade, 2015-12-16:
Ok, likely a firewall issue if if can't connect to the remote host.

sakshamgupta006, 2015-12-16:
I turned off the firewall as well before running , but the same error pops up...i am running this on windows 10.1

stephendade, 2015-12-17:
It could be that there's another program using the 5760 port. Try running sim_vehicle.sh -j4 -I1 --map. The "-I1" starts a second instance of SITL, using different ports.

Other than that, It's most likely a problem specific to your computer as I've been unable to replicate the issue on my Windows10 box and no other users have reported this issue.

icvdok, 2015-12-24:
Same problem here using arducopter on Windows10 cygwin

Here the output:
MAV> Exception in thread main_loop::127.0.0.1:5770
Traceback (most recent call last):
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\threading", line 810, in __bootstrap_inner
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\threading", line 763, in run
  File "<string>", line 717, in main_loop
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 374, in wait_heartbeat
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 339, in recv_match
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 302, in recv_msg
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.mavutil", line 885, in recv
error: [Errno 10053] An established connection was aborted by the software in your host machine

Firewalls deactivated. No active connections on port 5770 (netstat).

eleddie, 2016-01-07:
Did you solve the problem? I am getting the same error

sakshamgupta006, 2016-01-09:
I think the problem is due to windows 10.1 . The same thing worked on windows 8 in  virtual box.

icvdok, 2016-01-09:
Thanks for the feedback but in this case does make sense to setup a linux VM. Still wait for a native solution.

eleddie, 2016-01-09:
It's so annoying because I had it working like 3 days ago. 
I've tried disabling the firewall, reinstalling mavproxy, I've uninstalled some programs I installed after it was working and nothing..

sakshamgupta006, 2016-01-18:
Uninstall Shell&Services Engine if you have that on your system...
That worked for me....

closed 2016-01-19 by sakshamgupta006

YusufYigitbasi, 2018-06-30:
I am using linux ubuntu and i started to make project with dronekit-sitl mavlink mavproxy.Then while i trying to simulate simple_go_to.py script,which is  published tutorial video on the ardupilot's web.when i entered the out coordinates with this code on the terminal "mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551 --map --console " i take this error.

WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
STABILIZE> Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/yusuf/.local/bin/mavproxy.py", line 783, in main_loop
    master.wait_heartbeat()
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 383, in wait_heartbeat
    return self.recv_match(type='HEARTBEAT', blocking=blocking)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 344, in recv_match
    m = self.recv_msg()
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 318, in recv_msg
    msg = self.mav.parse_char(s)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9257, in parse_char
    self.__callbacks(m)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9234, in __callbacks
    self.callback(msg, *self.callback_args, **self.callback_kwargs)
  File "/home/yusuf/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_link.py", line 460, in master_callback
    mavutil.mavlink.MAV_TYPE_DODECAROTOR]:
AttributeError: 'module' object has no attribute 'MAV_TYPE_DODECAROTOR'

And my map didnt loaded and console type was unknown type.
then i ran simple_go_to.py with another terminal,but it gave timeout 30s error.
NOTE: my pip version is 8.0.1 python :: 2.7
What should i do?
Thanks for help

peterbarker, 2018-07-04:
On Sat, 30 Jun 2018, YusufYigitbasi wrote:

> AttributeError: 'module' object has no attribute 'MAV_TYPE_DODECAROTOR'

You need a more recent version of pymavlink.  If you've used pip to 
install it - use that to upgrade it :-)