ArduPilot/MAVProxy #252: python quit unexpectedly when loading modules on mac

antonvh, 2016-02-08:
I tried 'module load map' and 'module load console'. My mac says 'Python quit unexpectedly'. Running mavproxy as superuser doesn't help. Opencv and wx seem to work as i can import them in python and run hello world examples. Do I need pygame too?

stephendade, 2016-02-09:
Pygame is required for the joystick module. It shouldn't affect the console or map modules, but could be worth giving it a go.
Is it just any of the GUI modules that are affected?

antonvh, 2016-02-09:
What is a module that has no GUI and is not loaded from the start?

peterbarker, 2016-02-10:
On Tue, 9 Feb 2016, antonvh wrote:

> What is a module that has no GUI and is not loaded from the start?

dataflash_logger

So "module load dataflash_logger".

Note that "module load" will tab-complete after you give the first letter 
of the module name.

antonvh, 2016-02-12:
That one works. It's something to do with the GUI modules. Today there was a little progress: the python rocket showed in the Dock, but no console in sight. I haven't changed a thing. Will do some more digging tomorrow.

DarylWM, 2016-03-14:
`module load misseditor` doesn't work on OS X (10.11.3) either. I get the rocket in the dock but no window.

smiltee, 2017-04-27:
Still happening, April 2017
"load module map" results in a forever Python icon bouncing in the dock
OSX 10.12, mavproxy 1.6.1 from pip

peterbarker, 2017-04-28:
On Thu, 27 Apr 2017, smiltee wrote:

> Still happening, April 2017
> "load module map" results in a forever Python icon bouncing in the dock
> OSX 10.12, mavproxy 1.6.1 from pip

We have patches in master that should make things work on MacOSX.

If you are in a position to help testing that those patches work, that 
would be excellent... otherwise, sorry, you'll have to wait for the next 
release!

Peter

amilcarlucas, 2018-10-17:
The latest release should work

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas