ArduPilot/MAVProxy #254: windows: --dialect commands lead to errors

alexcherpi, 2016-02-23:
Hi,

I installed the version 1.4.40 of Mavproxy.
I have an error when I want to run the command '--dialect=common'.

Before running this command I generated and installed the mavlink python files by runnning 'setup.py install' (for mavlink). I moved the generated common.xml and common.py in 'C:\Program Files (x86)\MAVProxy\message_definitions\v1.0'. But still I have the following output error.

Here is the output error:

C:\Users\lis>mavproxy --dialect=common
Traceback (most recent call last):
  File "<string>", line 884, in <module>
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.mavutil", line 103, in set_dialect
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.mavgen", line 160, in mavgen_python_dialect
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.mavgen", line 53, in mavgen
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.mavgen", line 47, in mavgen_validate
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.lib.minixsv.pyxsval", line 105, in parseAndValidate
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.lib.minixsv.pyxsval", line 133, in parseAndValidateXmlInpu
t
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.lib.minixsv.pyxsval", line 230, in validateXmlInput
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.lib.minixsv.pyxsval", line 210, in parse
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\pymavlink.generator.lib.genxmlif.xmlifMinidom", line 83, in parse
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\urllib", line 87, in urlopen
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\urllib", line 210, in open
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\urllib", line 222, in open_unknown
IOError: [Errno url error] unknown url type: 'c'

What is wrong ?

stephendade, 2016-02-24:
> Before running this command I generated and installed the mavlink python files by runnning 'setup.py install' (for mavlink). I moved the generated common.xml and common.py in 'C:\Program Files (x86)\MAVProxy\message_definitions\v1.0'. But still I have the following output error.

You would have to run MAVProxy from source in order to pick up the new dialect, as pymavlink compiles the dialects at runtime. 

The MAVProxy from the windows installer run inside a minimal Python environment (independent from what other environments the user may have on their system) which isn't capable of compiling the other dialects at runtime.

alexcherpi, 2016-02-24:
Thanks for this information.
Do you think there will be soon a release of MAVProxy on windows which will take into account the command '--dialect' ?

stephendade, 2016-02-25:
In terms of an update to the installer, probably not for a while.
Otherwise, take a look at http://dronecode.github.io/MAVProxy/html/development/mavdevenv.html#windows for an alternate method of running MAVProxy on Windows (the --dialect should work here).

closed 2016-02-25 by stephendade