ArduPilot/MAVProxy #261: mavlogdump.py crashing on certain logs

AndKe, 2016-03-17:
I have several .BIN logs that cannot be parsed by mavlogdump

```
andre@Aranand:~$ mavlogdump.py /home/andre/Desktop/Flightlogs/err.BIN
Skipped 6299361 bad bytes in log (46, 83, 161) remaining=51660601
Traceback (most recent call last):
  File "/home/andre/.local/bin/mavlogdump.py", line 4, in <module>
    __import__('pkg_resources').run_script('pymavlink==1.1.62', 'mavlogdump.py')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 735, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1652, in run_script
    exec(code, namespace, namespace)
  File "/home/andre/.local/lib/python2.7/site-packages/pymavlink-1.1.62-py2.7-linux-x86_64.egg/EGG-INFO/scripts/mavlogdump.py", line 43, in <module>
    zero_time_base=args.zero_time_base)
  File "/home/andre/.local/lib/python2.7/site-packages/pymavlink-1.1.62-py2.7-linux-x86_64.egg/pymavlink/mavutil.py", line 1079, in mavlink_connection
    m = DFReader.DFReader_binary(device, zero_time_base=zero_time_base)
  File "/home/andre/.local/lib/python2.7/site-packages/pymavlink-1.1.62-py2.7-linux-x86_64.egg/pymavlink/DFReader.py", line 516, in __init__
    self.init_clock()
  File "/home/andre/.local/lib/python2.7/site-packages/pymavlink-1.1.62-py2.7-linux-x86_64.egg/pymavlink/DFReader.py", line 362, in init_clock
    m = self.recv_msg()
  File "/home/andre/.local/lib/python2.7/site-packages/pymavlink-1.1.62-py2.7-linux-x86_64.egg/pymavlink/DFReader.py", line 441, in recv_msg
    return self._parse_next()
  File "/home/andre/.local/lib/python2.7/site-packages/pymavlink-1.1.62-py2.7-linux-x86_64.egg/pymavlink/DFReader.py", line 582, in _parse_next
    null_term(elements[3]), null_term(elements[4]))
  File "/home/andre/.local/lib/python2.7/site-packages/pymavlink-1.1.62-py2.7-linux-x86_64.egg/pymavlink/DFReader.py", line 59, in __init__
    raise Exception("Unsupported format char: '%s' in message %s" % (c, name))
Exception: Unsupported format char: '' in message ,H�U
andre@Aranand:~$ 
```
An example can be found here: http://aranica.com/tmp/err.BIN

AndKe, 2016-03-17:
There's obviously something wrong with the file - I just don't know if something/more or less, should be expected to be usable anyway..

newts, 2018-08-23:
Was this on Windows?

I just found a bug in the .tlog files being corrupted on windows by newline translation.  Might be related, I noticed it with mavlogdump with corrupted messages, not crashing.

AndKe, 2018-08-23:
@newts no, I don't touch that stuff.

amilcarlucas, 2018-10-20:
There have been some fixes in this code, can you please retest with the latest version?

AndKe, 2018-10-20:
I dont have that log anymore, and now I see this issue is 2 years old, I use mavlogdump for processing rather often, and can say there is no such issue anymore.

closed 2018-10-20 by AndKe