ArduPilot/MAVProxy #262: MAVProxy Freezes up

Jesse-Millwood, 2016-03-22:
MAVProxy is freezing up after about 30 seconds to 6 minutes. 
I am able to press enter but commands are not executed, just a new line happens, but I can still see the prompt. I am also unable to exit the program with Ctrl-c. 

I am running MAVProxy like this:
```
mavproxy.py --master=/dev/ttyUSB0 --out=localhost:1456 --logfile=./mavlog --moddebug=3 --source-system=210 --quadcopter
```
After the parameters are loaded I run `watch HEARTBEAT` so I can tell that things are still happening. This is how I know it stops. 

I am running MAVProxy 1.4.42 on Ubuntu 15.10 (in virtualbox). I am communicating with an Iris+ running AC3.3.3. 

One odd thing though is that MAVProxy does not seem to freeze up when I'm connecting the Pixhawk directly to the computer. When I run `link` in MAVProxy it shows 0.2% loss here, where I was getting about 6% before.

peterbarker, 2016-03-22:
On Tue, 22 Mar 2016, Jesse Millwood wrote:

> MAVProxy is freezing up after about 30 seconds to 6 minutes.
> I am able to press enter but commands are not executed, just a new line
> happens, but I can still see the prompt. I am also unable to exit [...]

Could you do a "module list", please?

Jesse-Millwood, 2016-03-24:
@peterbarker sorry for the late response. I am away from my quad right now and won't be able to get to it until tomorrow. Thank you for the response!

Jesse-Millwood, 2016-03-25:
@peterbarker here is the output of my `module list` command
```
STABILIZE> module list
STABILIZE> link: link control
log: log transfer
link: link control
wp: waypoint handling
rally: rally point control
fence: geo-fence management
param: parameter handling
relay: relay handling
tuneopt: tuneopt command handling
arm: arm/disarm handling
mode: mode handling
calibration: calibration handling
rc: rc command handling
auxopt: auxopt command handling
misc: misc commands
cmdlong: cmdlong handling
battery: battery commands
terrain: terrain handling
output: output control
adsb: ADS-B data support
```

Jesse-Millwood, 2016-03-25:
Is it odd that link control is listed twice? Also the freezing up does not seem to happen with APM Planner 2 so I am not entirely convinced it is related to the virtual machine or Linux.

Jesse-Millwood, 2016-03-25:
well I'm going to close this and just not use a virtual machine for this. I used this same mavproxy command on a different laptop running Ubuntu 14.04 and mavproxy 1.4.40 (not in a vm) and it worked fine, I also tried mavproxy 1.4.42 on windows and it worked fine (also had the two link modules loaded).

closed 2016-03-25 by Jesse-Millwood