ArduPilot/MAVProxy #268: AssertionError: can only test a child process

larsks, 2016-04-26:
Attempting to use the map module results in lots of:

```
Traceback (most recent call last):
  File "/home/lars/src/ardupilot/.venv/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap_ui.py", line 437, in on_redraw_timer
    self.redraw_map()
  File "/home/lars/src/ardupilot/.venv/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap_ui.py", line 425, in redraw_map
    self.update_position()
  File "/home/lars/src/ardupilot/.venv/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap_ui.py", line 334, in update_position
    alt = self.ElevationMap.GetElevation(lat, lon)
  File "/home/lars/src/ardupilot/.venv/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_map/mp_elevation.py", line 41, in GetElevation
    tile = self.downloader.getTile(numpy.floor(latitude), numpy.floor(longitude))
  File "/home/lars/src/ardupilot/.venv/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 208, in getTile
    if childFileListDownload is not None and childFileListDownload.is_alive():
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 155, in is_alive
    assert self._parent_pid == os.getpid(), 'can only test a child process'
AssertionError: can only test a child process

```

It's not clear whether this is just noise or represents an actual problem.  The map eventually comes up.

DanMuratov123, 2016-05-01:
I have the same problem, can't find a solution .. 
Anyone ?

stephendade, 2016-05-01:
Which operating system are you running MAVProxy on?

DanMuratov123, 2016-05-05:
Linux Mint

larsks, 2016-05-05:
I'm using Fedora 23.

DanMuratov123, 2016-05-07:
larsks,

Does it work with Fedora 23 ? Do you have the same problem ? would you be
able to give an advice on how to fix the 'map' console problem ?

thank you,
Dan.

stephendade, 2016-05-15:
(Hopefully) done in #277. I'll close this when it's been merged in

mday299, 2016-06-17:
Issued confirmed here.

Just checked and the branch at #277 does appear to fix this issue.

mday299, 2016-06-17:
FYI: confirmed using Ubuntu 14.04.

stephendade, 2016-06-18:
@mday299 - is it the exact same error message as the original? Could you provide the error output you're getting?

mday299, 2016-06-18:
This one is proving exceedingly difficult to debug.  Once the map loads one time, the problem tends to go away, so to reproduce I have to start from a clean Ubuntu 14.04 VM.  Also, sometimes the issue causes the map to not load for several attempts, but on my previous attempt the map loaded after the 2nd try.  After the map loads once the problem is only an annoyance b/c the map seems to load every time after that.  

Anyway, the error I'm seeing _is_ somewhat different.  The only similarity is the 'can only test a child process' bit at the end:

```
Loaded module console
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
Process Process-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 93, in child_task
    self.app.frame = MPSlipMapFrame(state=self)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_map/mp_slipmap_ui.py", line 27, in __init__
    state.panel = MPSlipMapPanel(self, state)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_map/mp_slipmap_ui.py", line 228, in __init__
    self.ElevationMap = mp_elevation.ElevationModel()
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_map/mp_elevation.py", line 25, in __init__
    self.downloader.loadFileList()
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_map/srtm.py", line 109, in loadFileList
    self.createFileList()
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_map/srtm.py", line 115, in createFileList
    if childFileListDownload is None or not childFileListDownload.is_alive():
  File "/usr/lib/python2.7/multiprocessing/process.py", line 155, in is_alive
    assert self._parent_pid == os.getpid(), 'can only test a child process'
AssertionError: can only test a child process
STABILIZE> Unloaded module map
Received 623 parameters
Saved 623 parameters to mav.parm
```

stephendade, 2016-06-19:
It's a multithreading issue, with a bit of a race condition - depending on the state of 'childFileListDownload' at the time of execution. Hence the difficulty in replicating the issue.

peterbarker, 2016-06-20:
On Sat, 18 Jun 2016, Michael Day wrote:

> This one is proving exceedingly difficult to debug. Once the map loads one
> time, the problem tends to go away, so to reproduce I have to start from a
> clean Ubuntu 14.04 VM. Also, sometimes the issue [...]

I have recollections of side-stepping this issue when implementing 
mavproxy_firmware.py

mavproxy_firmware.py does much the same thing as srtm.py in terms of 
creating extra threads of execution for downloading *stuff*.

mavproxy_firmware.py uses threads rather than processes:

https://github.com/Dronecode/MAVProxy/blob/master/MAVProxy/modules/mavproxy_firmware.py#L342

vs

https://github.com/Dronecode/MAVProxy/blob/master/MAVProxy/modules/mavproxy_map/srtm.py#L116

stephendade, 2016-06-20:
@peterbarker - nice coding :) The issue is that the srtm file is called by multiple different modules in MAVProxy (there's a few modules that need terrain data). This was initially solved by having childFileListDownload as a global variable, so each thread can check if a different thread is already downloading the srtm filelist. Problem is that there's sometimes a race condition where childFileListDownload is being initialised by one thread, whilst read by another - leading to the aforementioned error message.

I was pretty sure I made it thread-safe with my latest commits to #277

tpascaru, 2016-08-27:
I am seeing this issue for the first time running MavProxy 1.4.38 on Ubuntu 16.04 running under HyperV on a windows10 machine. I am stuck on MavProxy 1.4.38 and am not sure if the problem was fixed in a later version (#277 above). Is there any workaround with my setup?

stephendade, 2016-08-29:
MAVProxy is currently at version 1.5.51, so try upgrading and see if that works

tpascaru, 2016-08-29:
Sadly, I'm stuck on v1.4.38 because we're using ArudPlilot 3.3.0 which is not compatible with later versions of MavProxy (under SITL) due to the fact that RC command overrides don't work (apparently, they stopped working when 16-channel support was added). If this issue is not addressable, we'll get by since we don't really need the MAP under Linux. I thought I'd bring it up though.

wynn4, 2016-10-03:
Same problem here on Ubuntu 14.04 laptop 'AssertionError: can only test a child process' Unloaded module map

I have MAVProxy installed on my desktop and it has no issues loading the map module

tridge, 2016-10-05:
I've pushed a fix into master for this issue
thanks for reporting it!

closed 2016-10-05 by tridge