ArduPilot/MAVProxy #275: some documentation upgrade

AndKe, 2016-05-11:
please update http://dronecode.github.io/MAVProxy/html/getting_started/download_and_installation.html#linux - 

1: the most recent version of the python-wxgtk library is python-wxgtk3.0 

2: just installing from source as described gives on serial connection mavproxy.py --master /dev/ttyUSB0) the awful error 'Serial' object has no attribute 'setBaudrate' -which is apparently solved by pip install --upgrade pymavlink  

That's hard to guess, as pymavlink package is not even mentioned in Linux instructions.
please consider adding a message that detects old package and suggests upgrade.

stephendade, 2016-05-15:
For 2), it turns out that the MAVProxy install did not have the correct versions of pymavlink, pyserial, etc referenced. Fixed in #279 . Once this goes in, MAVProxy will automatically download the correct versions of pymavlink and pyserial.

I'll fix 1) in a separate documentation update later.

stephendade, 2016-05-24:
1) is now fixed in d1897c8baad170630a35cb3a4cc689a3a2dc642b

stephendade, 2016-05-28:
Merged in

closed 2016-05-28 by stephendade