ArduPilot/MAVProxy #276: Why are garbage characters showed, and how to prevent ?

AndKe, 2016-05-11:
Over an radio link (with ECC), I sometimes see things like:

[image]

why ?  - and how to disable ?

magicrub, 2016-05-11:
Ouch, that doesn't look good. Do you have a tlog or bin log you can share?

AndKe, 2016-05-11:
unfortunately those logs was overwritten, but I've managed to reproduce the issue: 
hopefully, the attached .zip should look like the screenshot at your end too:
[image]

[mav.tlog.zip](https://github.com/Dronecode/MAVProxy/files/259996/mav.tlog.zip)

SamuelDudley, 2016-06-16:
There is some noise on your telemetry link and mavproxy is displaying the corrupted packets. You can hide it from appearing in the console by typing  `set shownoise false` as per the [docs](http://dronecode.github.io/MAVProxy/html/getting_started/faq.html):

> I have random “noise” displaying on the console, showing as corrupted text. How do I remove it?
>
>MAVProxy will, by default, show corrupted packets. To disable showing them, use `set shownoise false` in the MAVProxy console.

The corrupted packets indicate a low signal to noise ratio. I would suggest that you either have a strong local noise source near one of your telemetry antenna (video transmitter, other modem, etc...) or the telemetry signal is weak enough that a weak noise source is having an impact (poor antenna alignment, path loss is too great, de-constructive interference, etc...).

peterbarker, 2016-09-19:
The chances of a corrupt packet getting through are pretty slim, considering MAVLink's own CRC.  However, the chances of them getting as far as MAVLink's decoding aren't that bad.  My short amount of reading shows Golay12 will detect up to 7 bits of error in 24.  Which is fine, but I can see a standard telemetry link having more corruption than that.

Turning ECC off on the telemetry radios is not uncommon.

What you're probably seeing here are the bytes which do *not* correspond to valid packets being spat out.  The option @SamuelDudley pointed you at stop these unrecognised bytes being spat out.

@AndKe Are you happy with that explanation?

AndKe, 2016-09-19:
@peterbarker i guess you are right, it seems to happen more in my workshop (more RF noise) - also, the few characters that pop up at a time- can't be a mavlink packet, just some random data that passes golay.
Good answer - thank you.

closed 2016-09-19 by AndKe