ArduPilot/MAVProxy #278: RC override doesn't work with Copter-3.2.2

dricketts, 2016-05-15:
Support for 16 rc channels seems to break RC override when running Copter-3.2.2 using the SITL simulator. That is, commands of the form rc <chan> <value> do not work starting with this commit:

https://github.com/Dronecode/MAVProxy/commit/d81f4062f73fbe3ea608f66b767971f5c3306e98

I'm not currently able to test whether this same issue occurs on an actual vehicle. I also can't figure out where the issue is in ardupilot.

Finally, I'm not sure if this is a bug with Copter-3.2.2, MAVProxy, or not a bug at all, but for lack of a better place, I thought I'd report it here. I apologize in advance if this is the wrong place.

VermilianBIO, 2016-06-07:
This problem occurs in Copter-3.3.3 as well. However, the actual vehicle can be override and works just fine.

Pretty sure the problem is due to the newer version of MAVProxy changing the support rc channel to 16 channels while non-master ardupilot only support 8. I have tried downgrading my MAVProxy version back to 1.4.38 where it's still only has 8 rc channels and now it's working again.

dricketts, 2016-06-07:
Sorry, that was a typo. I meant Copter-3.3.2.

Downgrading the version works - thanks. I stupidly didn't realize that was possible.

closed 2018-10-20 by amilcarlucas