ArduPilot/MAVProxy #280: [feature request] Support python3

richerlariviere, 2016-05-21:
Do you plan to port MAVProxy to python 3?

stephendade, 2016-05-22:
It'd be awesome to do that. The only issue holding us back is that wxPython (which we use for our GUI) is itself not compatible with Python 3.

The non-gui parts of MAVproxy could definitely be ported. The pymavlink package (backend UAV protocol library) would need to be ported.

monkeypants, 2018-04-10:
wxPython 4.0 is out in BETA,  https://wxpython.org/

The pypi listing says it supports python 3.4, 3.5 and 3.6.

stephendade, 2018-04-11:
And there's a pull request at #420 to do exactly that. There were some concerns about breaking Python 2.7 compatbility though

amilcarlucas, 2018-10-17:
Try the latest version.

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas