ArduPilot/MAVProxy #281: GUI modules don't always work on Mac OS X 10.11.4

staroselskii, 2016-05-23:
It seems like WX related code stopped functioning properly some time ago.

**Test environment:**
- OS: 10.11.4
- python: 2.7.11
- wxpython: 3.0.2.0

**Steps to reproduce:**
- launch mavproxy.py
- Wait for a connection
- ```module load console```

After that Python icon screams for help in the Dock without showing a window.

I had't noticed this issue until I tried to dynamically ```module load console``` or ```module load map```. Everything seems to be working fine when I pass ```--console``` or ```--map``` respectively on the command line.

A quick look at the code hasn't yielded any results yet. Can it have anything to do with #163?

I'm also eager to know if this issue concerns only me or everyone on Macs?

Everything works fine using the same version of wxPython on Arch Linux.

stephendade, 2016-05-24:
There are some compatibility issues with MAVProxy on Macs - nearly all to do with wxPython.

Unfortunately, none of the developers have access to a Mac for development/debugging. So it's not something we're able to fix :(

prat96, 2016-05-27:
Lots of bugs running MAVProxy on OSX.. looks like I will have to install linux on my Macbook :/

amilcarlucas, 2018-10-17:
Try the latest version.

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas