ArduPilot/MAVProxy #284: replace the MAV_DATA_STREAM with MESSAGE_INTERVAL

kralo, 2016-05-24:
MAV_DATA_STREAM is deprecated as of July 15 ( [Common Message ver3](https://pixhawk.ethz.ch/mavlink/) ) which is used in set_stream_rates() .

peterbarker, 2016-09-19:
That looks interesting.  A couple of queries:
 - does anything support MESSAGE_INTERVAL?  Ground stations?
 - how does one transition from MAV_DATA_STREAM to MESSAGE_INTERVAL?  Can they co-exist?
 - what happens if you have a mixed network with some GCS asking for MAV_DATA_STREAM and some for MESSAGE_INTERVAL?
 - how does one efficiently pack the information on an autopilot for this?  (this one's more of a rhetorical question ;-) )

amilcarlucas, 2018-10-17:
This needs a big PR to ardupilot, and some changes to mavproxy.
The good news is that Peter is working on it