ArduPilot/MAVProxy #303: error using sim_vehicle.py -w showing c\users\Stephen fixed path

magicrub, 2016-08-17:
I don't know what version of MAVproxy I'm running, is there a way to check? The file is dated June 15, 2016 which matches 1.5.0 so I'm assuming that's it.
I ran command: `sim_vehicle.py -j4 -S10 -w` and this popped up.
I would have brushed it off except it showed the error in the fixed path of:
C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py

[image]

stephendade, 2016-08-18:
To get the current version:
```
module load help
mavhelp about
```

The references to "C:\Users\Stephen\Documents\MAVProxy" are expected. It's a side effect of when I compile MAVProxy on my computer.

peterbarker, 2016-09-19:
@magicrub is this repeatable?

The exception looks genuine - it tried to open a file (at a seemingly reasonable location (for Windows...)) and failed.  Have you previously had a MAVProxy installed which you've manually deleted or something?

You could always try reinstalling...

If the exception doesn't happen any more then we could probably close this issue as @stephendade has explained it's an artifact of the build process for Windows.

stephendade, 2017-03-18:
Fixed in MAVProxy 1.6.1

closed 2017-03-18 by stephendade