ArduPilot/MAVProxy #304: Ask for more docs on how to write a mavproxy script on our own

Zhuoran-Zhang, 2016-08-22:
I wrote a very simple script: 
motortest 1 0 30 2
motortest 2 0 30 2
but my Pixhawk reads only one command. I also tried different ways to seperate those two commands but it doesn't work either. Unfortunately, I couldn't find some more informations about how to write Mavproxy scripts. There are only some mavinit.scr files which are hard to use for a simple test.
So I would like to know if there are some tutos for correctly writing a script?

magicrub, 2016-08-22:
This is the feature request and bug tracking system. For help try discuss.ardupilot.org

Zhuoran-Zhang, 2016-08-22:
Thanks for mentioning and sorry for the inconvenience

peterbarker, 2016-09-19:
@Zhuoran-Zhang the commands you entered simply send a mavlink packet to the vehicle and return.  The craft would happily have ignored the second command while it was doing the first.

To get synchronous behaviour you will need to look at writing a module.  An example module has recently been committed, which you might find useful as a starting point.

Since this isn't an actionable FR or bug I'm going to close this issue, and like @magicrub I suggest you try discuss if you need further help.

closed 2016-09-19 by peterbarker