ArduPilot/MAVProxy #307: Mavlink/MavExplorer on OSX

lvale, 2016-08-24:
trying to run mavexplorer on OSX and when starting this comes up

```
Sparkle:~ Luis$ MAVExplorer.py /Users/Luis/apmplanner2/dataflashLogs/00-01-01_00-13-06.bin 
Aug 24 21:32:32  Python[3535] <Error>: Set a breakpoint at CGSLogError to catch errors as they are logged.
Aug 24 21:32:32  Python[3535] <Error>: void CGSGetDisplaySystemState(uint64_t, CGSDisplaySystemState **): MIG error 0x10000003: (ipc/send) invalid destination port
This program needs access to the screen.
Please run with a Framework build of python, and only when you are
logged in on the main display of your Mac.

MAV> 
MAV> 
```

Some guidance to troubleshoot would be welcome.

The required python libs are installed and also X11.

lvale, 2016-08-24:
Perhaps this https://github.com/ArduPilot/MAVProxy/issues/281 is the same/related ?

stephendade, 2016-08-27:
Yep, same issue as the one you linked above. Might be something to deo with this: https://groups.google.com/a/continuum.io/forum/#!topic/anaconda/_envRc_iy7Q

lvale, 2016-08-28:
so, the dependency on wxwidgets is killing this ? Or is there a way to invoke pythonw like referenced on your link.

stephendade, 2016-08-29:
The major issue is that none of the MAVProxy developers have access to an OSX computer, so we're unable to debug or fix the issue.
If you're able to investigate it, that would be much appreciated - maybe start with the link I posted above :)

closed 2017-01-02 by lvale