ArduPilot/MAVProxy #327: Issue with MAVProxy 'position' command in GUIDED mode

dc9500, 2016-10-26:
When operating in GUIDED mode, issuing the command 'position x y z' invokes the cmd_position()
method in [mavproxy_cmdlong.py](https://github.com/ArduPilot/MAVProxy/blob/f35a9837f1d6861fd558c9f90171ac6e2af8e251/MAVProxy/modules/mavproxy_cmdlong.py#L229-L249)

Observe these lines of the function in particular:

```
            self.master.mav.set_position_target_local_ned_send(
                                      0,  # system time in milliseconds
                                      1,  # target system
                                      0,  # target component
                                      8,  # coordinate frame MAV_FRAME_BODY_NED
                                      3576,     # type mask (pos only)
                                      x_m, y_m, z_m,  # position x,y,z
                                      0, 0, 0,  # velocity x,y,z
                                      0, 0, 0,  # accel x,y,z
                                      0, 0)     # yaw, yaw rate
                                      0, 0, 0,  # accel x,y,z
                                      0, 0)     # yaw, yaw rate
```
Why is 'target system' hardcoded to 1 in cmd_position() (and cmd_velocity() for that matter), when all other command definitions use this.settings.target_system instead? The consequence I have observed in not using 'this.settings.target_system' is that MAVProxy's 'position' and 'setspeed' commands are simply not relayed to an autopilot with a SYSID_THIS_MAV parameter of greater than 1.

I have resolved the issue through replacing '1' with 'this.settings.target_system' in my own local copy of mavproxy_cmdlong.py and have eliminated the issue.

mtrivedi25, 2020-08-12:
What setting did you replace it with? I am having a similar issue in my custom module.

peterbarker, 2020-09-22:
I've created a PR to address this.

closed 2020-09-22 by peterbarker