ArduPilot/MAVProxy #334: Adding RPi GPIO switch functionality

JAR4x4, 2016-10-31:
Hello,

I don´t know if this is the right place to post a question like this, but I´ll try.

I´m new to Python and programming in general, but I´m learning.

Currently I´m working on a project with a standalone GCS unit based on two Raspberry Pi´s in a Peli case together with two 7" Raspberry touchscreens. One for the telemetry and one for a video downlink. (I´m using a RPi with Navio2 as flightcontroller).

I´ve successfully got the "telemetry" RPi with a custom USB joystick interface up and running with MavProxy and communicating with my Navio2. Now I´m studying the MavProxy code to find a way to issue some predefined terminal commands like "arm throttle" "rtl" etc with switches connected to the RPi GPIO or with my USB joystick interface by mapping the buttons to trigger a standalone script to do the tasks I want. 

Basically I want to trigger the mavproxy_arm.py and mavproxy_mode.py with switches connected to the RPi.

I´m stuck. Can anyone point me in the right direction?  

Best regards
Alex from Norway

stephendade, 2016-11-01:
Hi Alex,

There's a MAVProxy forum on ardupilot.org: http://discuss.ardupilot.org/c/ground-control-software/mavproxy

That would be the best place to post your question.

monkeypants, 2016-11-01:
Yes, a question for the forum... However (sorry, I really should be answering there)

If you have are reading your switches from sucessfully from the rpi GPIO already, then you might try making a custom mavproxy module to respond to them from within MavProxy.

http://ardupilot.github.io/MAVProxy/html/development/mavcustmod.html

Look at the joystick module (for example), see how idle_task method is defined to poll the joystick for input (and send 'rc' module overrides in response to joystick input). You could do something like that with the rpi.GPIO too, but you will have to dig a little deeper to figure out exactly what message you want your switches to send.

JAR4x4, 2016-11-01:
@stephendade thanks for the reply, I will remember that for future questions. Btw I was not able to create an account for the forum..

@monkeypants thanks for your input. I started some work on a custom module last night. I´ve managed to recognize the rip.GPIO inputs in the module. When i push the button the module starts self.master.arducopter_arm() just like when you issue the command "arm throttle".

I don´t remember the exact error message I get when I press the button now, maybe I can send you a private message with my code and error message when I get home so that we can close this thread?

Alex

monkeypants, 2016-11-01:
I think maybe just close this ticket and write your message in the forum - open tickets here are basically "action required" on software maintenance - like bugs and feature requests etc. I just joined the forum myself and it all seems to be working.

closed 2016-11-01 by JAR4x4