ArduPilot/MAVProxy #343: Changing parameter from `--cmd` switch on start

Jesse-Millwood, 2016-11-07:
I am trying to figure out if this is a problem with MAVProxy or sim_vehicle.py but I think it is an issue with MAVProxy. 
When MAVProxy gets a string for the `--cmd` switch it runs `module load` just fine but if you do something with the params it will say `Unable to find parameter 'PARAMETERNAME'`. Then if you run the same command once it has connected to the vehicle, it will work. 
I have started a simulated instance with the output `udp:127.0.0.1:14560` and then ran 
```
mavproxy.py --master=udp:127.0.0.1:14560 --cmd="param set SYSID_THISMAV 3"
```
and it says `Unable to find parameter 'SYSID_THISMAV'`. But if I run that command after it connects it is fine. So it seems there is an issue with how MAVProxy handles the `--cmd` switch on start. 
Should it wait to run the commands it gets in the `--cmd` switch until it gets heartbeats?

aoigetei, 2017-12-20:
Hi, 
I have run into the same issue. Did you ever figure out a way to get this to work ?

faxik, 2019-06-05:
Hit me as well. Any progress on that?

Jesse-Millwood, 2019-06-05:
At the time I was using this, I believe I just hacked up how the arguments were parsed for `cmd`. I think I made a pull request, which worked, but as I said was a little hacky. I'll try and see if I can find that.

peterbarker, 2019-06-11:
Just throwing `param set`s around in `--cmd` really isn't sufficient - it's going to be unreliable even if you wait for heartbeats.

Some sort of simple module may be appropriate - we have a mechanism for passing parameters through to those IIRC.