ArduPilot/MAVProxy #357: Graph module's left side scale is odd if value is constant for too long

rmackay9, 2016-12-19:
[image]

In the 2nd picture above you can see how the scale on the left always shows "0.10" if the value doesn't change for a while.

stephendade, 2016-12-26:
Yep, that's the exact behavior according to lines 108-110 of MAVProxy/modules/lib/live_graph_ui.py

If the graph line is a constant value, the y-axis limits go to +-0.1 around the constant value. Really need a method to have smarter y-axis limits

rmackay9, 2016-12-26:
Thanks.  It actually prints the wrong value though as well.  I could live with it squeezing down to just a small area around the value.. but look at the value shown!  It doesn't show 1850, instead it show 0.1... a completely unrelated value (as far as I can tell it's unrelated anyway).

stephendade, 2016-12-27:
It's the graph formatting - there's a exponent (partially) hiding behind the legend. Numbers are correct, just badly formatted. I've fixed it a #361, just have to get @tridge to merge it in.

stephendade, 2016-12-28:
Has been committed to master.

closed 2016-12-28 by stephendade

rmackay9, 2016-12-28:
Thanks very much!