ArduPilot/MAVProxy #358: Apm 2.8 won't arm

fbonin5491, 2016-12-19:
it will not arm , even if i disable the prearm-check, the channels seems to be corrects when i calibrate the receiver...any idea?I had an issue before with a Naze32 , my minimum throttle was not low enough, could it be the same?

stephendade, 2016-12-19:
APM 2.8 is a very old version ... is there a particular reason you're not using the latest APM software?

fbonin5491, 2016-12-19:
Thanks for fast reply,
i don't think its a very old *hardware* , and it is the one i bought with
M8N gps
+compas.Accordinng to the software/firmware , i am using AC 3.4.1

rmackay9, 2016-12-19:
Support questions should really go into the forums so I'll close this if that's ok.  http://discuss.ardupilot.org/

By the way, if you have a ground station attached, normally the reason is displayed.

closed 2016-12-19 by rmackay9

fbonin5491, 2016-12-19:
my ground station is Mission Planner , and says nothing, except Disarmed.I dont  understand why you close this topic so fast.

rmackay9, 2016-12-19:
@fbonin5491,
Support issues should go into the support forum (linked above).  Sorry to be strict but if we allow support items to be raised in our issues list, then it just becomes another support forum and it hides the confirmed issues and to-do that we need to resolve.

rmackay9, 2016-12-19:
@fbonin5491, just one thing about the hardware.  If it's an APM board then it's a really old *design*.  Some hardware vendors are still selling these but the highest firmware they can run is AC3.2.1 which is nearly 2 years old.  Although the MP may show AC3.4.3 as the latest version available, when the firmware is uploaded you should see a message saying that it's actually AC3.2.1 that's been written to the board.

I don't know why manufacturers are still selling the APM boards.  They should at least put a warning on their sites saying the Pixhawk is a better option.

fbonin5491, 2016-12-19:
ok , understood , any way , my problem was my transmitter not going far
enough on the yaw channel, i push the trim to the right and was able to arm
the board.the pixhawk is also more expensive no?

rmackay9, 2016-12-19:
@fbonin5491, ah, great that it's sorted.
There are lots of very reasonably priced Pixhawk clones.  Of course I recommend buying from one of the stores that supports ardupilot. http://ardupilot.org/ardupilot/docs/common-stores.html