ArduPilot/MAVProxy #360: –default-modules cannot override .mavinit.scr to disable console module

tribbloid, 2016-12-22:
console module is not useful in command-line environment and many integrated system. However, so far the only way to reliably disable it is to remove 'module load console' from .mavinit.scr.

If not removed the console module will always be loaded regardless of any mavproxy.py command-line option, even with the new –default-modules it is not possible:

```
mavproxy.py --default-modules='link' --daemon
```

This will still pop up the console Window on my linux + Gnome. Haven't test on Windows yet.

Is this a deliberate design or should be enabled?

peterbarker, 2016-12-23:
On Thu, 22 Dec 2016, Peng Cheng wrote:

> If not removed the console module will always be loaded regardless of any
> mavproxy.py command-line option, even with the new –default-modules it is
> not possible:

> Is this a deliberate design or should be enabled?

Probably not a deliberate design decisoin, but the way it should work.

We probably need a "--no-mavinit-scr" option to MAVProxy; this will aid 
when running autotest.py, for example, as having the ex console pop up 
swallows the output before pyexpect sees it.

tribbloid, 2016-12-24:
Even better, in fact, the reason I propose it is exactly for easier integration test with SITL.

peterbarker, 2016-12-24:
@tribbloid Better to think positively.  This new option in the related PR allows you to specify a replacement for the script rather than just saying not to use the default value.

Please test!

tribbloid, 2016-12-26:
@peterbarker Manually tested on Ubuntu 16.04 + Python 2.7.12 using:

mavproxy.py --start-script "" --default-modules "link"
PASSED! initialization of console module suppressed as intended.