ArduPilot/MAVProxy #363: High CPU usage

rodizio1, 2017-01-10:
is there a way to lower CPU usage for mavproxy.py?

It uses about 10% of a CPU core on a Raspberry Pi3 when doing nothing, i.e. no telemtry being transferred and goes even higher when telemetry is being transferred.

Basically, I just need the forwarding from serial to UDP feature, is there a way to disable unneeded functions like logging or whatever other stuff it does?

peterbarker, 2017-01-10:
On Tue, 10 Jan 2017, Rodizio wrote:

> is there a way to lower CPU usage for mavproxy.py?
>
> It uses about 10% of a CPU core on a Raspberry Pi3 when doing nothing, i.e.
> no telemtry being transferred and goes even higher when telemetry is [...]

Nobody has really worked on reducing the CPU usage, past ensuring that it 
is using the C code for encoding/decoding.

Part of the issue is also one of the great things about mavproxy.py - the 
separate-process model used to increase stability in the face of bugs in 
mavproxy; freezing/unfreezing objects as they go into the other processes 
is expensive.

The AP_Sync project is currently using cmavnode to provide a simple proxy: 
https://github.com/peterbarker/companion/blob/clean-setup-scripts/RPI2/Raspbian/6_setup_cmavnode.sh

mavproxy.py can then connect to cmavnode if desired.

rodizio1, 2017-01-12:
Thanks for the info. Using cmavnode now, much lower CPU usage :)

fnoop, 2017-02-09:
I second this emotion - mavproxy is a fantastic system but the cpu usage is a killer on low power computers.  A single instance uses 35-40% cpu on a raspberry pi 2 connected to an idle SITL session.

RichLewis007, 2018-02-27:
I'm running MAVProxy on a Raspberry Pi Zero W and with no GUI, in CLI boot mode I'm seeing 71% CPU usage with the top monitoring app in Raspbian.  With the GUI loaded it hivers in the 90% range and up, just connected to a Pixhawk sitting next to it. 

Could I have some MAVLink debug messages streaming, or some other setting to cause this high CPU load? Or is the Pi Zero that much slower than the Pi 3 rodizio1 mentioned previously?

RichLewis007, 2018-02-27:
I see a discussion of this here: https://discuss.ardupilot.org/t/how-to-reduce-cpu-usage-of-mavproxy/12032

amilcarlucas, 2018-10-17:
Can you retry with the latest version?

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas