ArduPilot/MAVProxy #373: `magical_ui` unable to find its `.obj` files

courtarro, 2017-02-06:
MAVProxy is installed to /usr/local...

After starting MAVProxy, I ran `module load magical`, then `magical_ui`. The window appears, but the OpenGL areas are not rendered. In the console, this is the error:

    Traceback (most recent call last):
      File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
        self.run()
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/wavefront.py", line 172, in run
        self.obj = self.parser.parse(progress_callback=self.progress_callback)
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/wavefront.py", line 55, in parse
        return self.parse_file(progress_callback)
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/wavefront.py", line 96, in parse_file
        s = os.stat(self.filename)
    OSError: [Errno 20] Not a directory: '/usr/local/lib/python2.7/dist-packages/MAVProxy-1.5.7-py2.7.egg/MAVProxy/modules/mavproxy_magical/data/quadcopter.obj'

    Traceback (most recent call last):
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 143, in CreateRenderer
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 59, in __init__
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/wavefront.py", line 55, in parse
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/wavefront.py", line 96, in parse_file
    OSError: [Errno 20] Not a directory: '/usr/local/lib/python2.7/dist-packages/MAVProxy-1.5.7-py2.7.egg/MAVProxy/modules/mavproxy_magical/data/arrow.obj'
    Traceback (most recent call last):
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 143, in CreateRenderer
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 59, in __init__
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/wavefront.py", line 55, in parse
      File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/wavefront.py", line 96, in parse_file
    OSError: [Errno 20] Not a directory: '/usr/local/lib/python2.7/dist-packages/MAVProxy-1.5.7-py2.7.egg/MAVProxy/modules/mavproxy_magical/data/arrow.obj'

peterbarker, 2017-02-07:
On Mon, 6 Feb 2017, Ethan Trewhitt wrote:

> MAVProxy is installed to /usr/local...

How did you install MAVProxy?

Does map suffer from the same problem?

mday299, 2017-02-07:
I work with @courtarro -- we install MAVProxy using the setup.py file on Ubuntu 16.04 as follows:

python setup.py build install

I found the same issue and use the following workaround (git diff output -- basically I had to hard code the paths to the obj files because the "dirname" command couldn't find them in the .egg file):

> > diff --git a/MAVProxy/modules/mavproxy_magical/magical_ui.py b/MAVProxy/modules/mavproxy_magical/magical_ui.py
> index 8bc0bb0..28657d5 100644
> --- a/MAVProxy/modules/mavproxy_magical/magical_ui.py
> +++ b/MAVProxy/modules/mavproxy_magical/magical_ui.py
> @@ -604,7 +604,8 @@ class MagicalFrame(wx.Frame):
>          self.main_panel.GetSizer().Fit(self)
>  
>          p = os.path
> -        path = p.join(p.dirname(__file__), 'data', 'quadcopter.obj')
> +        #path = p.join(p.dirname(__file__), 'data', 'quadcopter.obj')
> +        path = p.join('/home/michael/ACS/mavproxy/MAVProxy/modules/mavproxy_magical', 'data', 'quadcopter.obj')
>          self.vehicle_loader = wv.ParserWorker(
>              wv.ObjParser(filename=path, enable_cache=True),
>              complete_callback=self.VehicleLoadCompleteCallback,
> diff --git a/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py b/MAVProxy/modules

/mavproxy_magical/wxgeodesicgrid.py
> index 8cd656f..ab31000 100644
> --- a/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py
> +++ b/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py
> @@ -55,7 +55,8 @@ class Renderer(glrenderer.Renderer):
>          self.vehicle = None
>  
>          p = os.path
> -        path = p.join(p.dirname(__file__), 'data', 'arrow.obj')
> +        #path = p.join(p.dirname(__file__), 'data', 'arrow.obj')
> +        path = p.join('/home/michael/ACS/mavproxy/MAVProxy/modules/mavproxy_magical', 'data', 'arrow.obj')
>          obj = wv.ObjParser(filename=path).parse()
>          self.mag = opengl.WavefrontObject(obj)
>          self.mag.local.scale(.88)
>

SamuelDudley, 2017-02-07:
Pretty sure I had the same issue with the cesium module... 
Perhaps adding `package_data.extend(package_files('MAVProxy/modules/mavproxy_magical/data'))` to setup.py would solve the issue?

peterbarker, 2017-02-07:
@guludo thoughts?

guludo, 2017-02-08:
@peterbarker @courtarro @mday299 

Hello guys,

Sorry for the trouble this has caused :-) I should have thought about package distribution when implementing the caching mechanism. My bad!

That being said, the reason I implemented that caching mechanism was because the module was taking too much time to parse the object file for the quadcopter. In the end I figured out that the reason for the problem was that the object had way too many (really) vertices and I asked Ronaldo (the guy who authored the model) to make one with less vertices. After that the parsing time decreased **significantly**! But, since I had already implemented the caching mechanism, I decided to leave it there. However I wasn't very smart about where to store the cached data :-P

Thus, I believe we don't really need caching for now and I think just removing the caching logic would solve this and #374.

If you guys agree on that, I'll try to do this by the end of the week. :-)

peterbarker, 2017-02-08:
@guludo No need for us to agree - you're obviously maintainer for the module and so long as it doesn't impact other places - you're free to make whatever improvements you like :-)

Thanks very much for doing this - people love the bling!

courtarro, 2017-02-08:
@guludo No problem! Will the caching change also fix the failure to find the `.obj` files? I would have assumed they would be necessary no matter what.

guludo, 2017-02-08:
@courtarro You`re right. I posted the answer for the wrong question. In fact, I just scammed this and thought it was the same as #374.

(in my defense, almost 1am here in Brazil and my concentration level has decreased haha just kidding)

@SamuelDudley's solution sounds interesting. I'll take a look on this as well.

guludo, 2017-02-20:
Hi guys, sorry about the delay... I had some issues and had to postpone this fix for a while.

@courtarro Could you please check if this https://github.com/ArduPilot/MAVProxy/pull/383 works for you? It worked from my environment.

closed 2017-03-14 by tridge