ArduPilot/MAVProxy #374: magical_ui unable to run when installed to system as a wheel

courtarro, 2017-02-07:
In an attempt to explore #373, I built and installed MAVProxy as a wheel instead of an egg. I followed these steps:

    user:~/mavproxy$ python setup.py bdist_wheel
    user:~/mavproxy$ cd dist/
    user:~/mavproxy$ sudo pip install MAVProxy-1.5.7-py2-none-any.whl
    user:~/mavproxy$ mavproxy.py

After MAVProxy loads, enter:

    module load magical
    magical_ui

And here's the set of errors I get:

    Traceback (most recent call last):
      File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
        self.run()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 172, in run
        self.obj = self.parser.parse(progress_callback=self.progress_callback)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 55, in parse
        return self.parse_file(progress_callback)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 121, in parse_file
        progress_callback=progress_callback,
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 152, in parse_lines
        self.parse_line(line, obj)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 269, in parse_line
        l = parser.parse()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 55, in parse
        return self.parse_file(progress_callback)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 125, in parse_file
        with open(self.filename + '.deps', 'w') as deps_file:
    IOError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/data/quadcopter.mtl.deps'

    Traceback (most recent call last):
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
        self.CreateRenderer()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 143, in CreateRenderer
        background=(r / 255.0, g / 255.0, b / 255.0, 1),
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 59, in __init__
        obj = wv.ObjParser(filename=path).parse()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 55, in parse
        return self.parse_file(progress_callback)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 121, in parse_file
        progress_callback=progress_callback,
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 152, in parse_lines
        self.parse_line(line, obj)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 269, in parse_line
        l = parser.parse()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 55, in parse
        return self.parse_file(progress_callback)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 125, in parse_file
        with open(self.filename + '.deps', 'w') as deps_file:
    IOError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/data/arrow.mtl.deps'
    Traceback (most recent call last):
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
        self.CreateRenderer()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 143, in CreateRenderer
        background=(r / 255.0, g / 255.0, b / 255.0, 1),
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 59, in __init__
        obj = wv.ObjParser(filename=path).parse()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 55, in parse
        return self.parse_file(progress_callback)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 121, in parse_file
        progress_callback=progress_callback,
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 152, in parse_lines
        self.parse_line(line, obj)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 269, in parse_line
        l = parser.parse()
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 55, in parse
        return self.parse_file(progress_callback)
      File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wavefront.py", line 125, in parse_file
        with open(self.filename + '.deps', 'w') as deps_file:
    IOError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_magical/data/arrow.mtl.deps'

In other words, magical is trying to write to files in its path, but that doesn't work because it's installed to the system. Bad idea.

peterbarker, 2017-02-07:
@guludo Can we use a mkstemp-or-equivalent function here instead?

guludo, 2017-02-08:
Hi @peterbarker 

I believe I might have an even easier solution hehe 

https://github.com/ArduPilot/MAVProxy/issues/373#issuecomment-278208778

By the way, sorry about the hassle :-)

guludo, 2017-02-20:
Hi guys, sorry about the delay... I had some issues and had to postpone this fix for a while.

@courtarro I disabled caching here: https://github.com/ArduPilot/MAVProxy/pull/382.

courtarro, 2024-01-09:
No longer relevant since caching is disabled. Closing.

closed 2024-01-09 by courtarro