ArduPilot/MAVProxy #379: Installation trouble with "wheel"

AndKe, 2017-02-12:
Fresh install on Ubuntu 16.04 (LTS):
- and a second run find nothing to do..

[image]

[image]

khancyr, 2017-02-15:
@AndKe It is known pymavlink problem... But the install should be find even if the first build step failed for pymavlink. The install is good so everything should work as expected.
I am award of this problem... I am sorry I didn't find the time to work on that yet ... The issue is that pymavlink is building the mavlink header at the pip build phase and need future and lxml for that. Even if pip dl them they are not available at build phase, so it need to wait the install phase to success !

AndKe, 2017-02-15:
ok, good to know, yes, as far I could test it worked fine, but I was not
sure if there would be some future problems due to that.

amilcarlucas, 2018-10-20:
try `python setup.py sdist` and then installing the resulting .whl file.

AndKe, 2018-10-20:
Yes, I haven't seen this issue since then, it is no longer an issue.

closed 2018-10-20 by AndKe