ArduPilot/MAVProxy #390: Magical - OpenGL issues

stephendade, 2017-03-18:
Running "module load magical" and the "magical_ui" gives the following errors:

(Note I'm running APM:Plane in SITL, Ubuntu 16.04)

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxvehicle.py", line 79, in CreateRenderer
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxvehicle.py", line 33, in __init__
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 34, in __init__
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/opengl.py", line 601, in compile_and_link
Exception: Error on compiling fragment shader: 0:2(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 143, in CreateRenderer
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 34, in __init__
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 34, in __init__
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/lib/opengl.py", line 601, in compile_and_link
Exception: Error on compiling fragment shader: 0:2(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES
```
My laptop (Dell Inspiron 3330) doesn't support the latest version of OpenGL. Is it possible for Magical to require a lower version of OpenGL? It would allow a wider variety of computers to run it.

stephendade, 2017-03-18:
@guludo

guludo, 2017-04-24:
Hi! I haven't had time to look at this issue yet. Sorry. I will take a look, but I'm afraid it won't be anytime soon. If someone is up for the task, I'd be glad to help with what I can. :-)