ArduPilot/MAVProxy #396: can't load mission editor on windows

tridge, 2017-03-26:
get this when loading editor on windows:

  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 649, in save_dict
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 681, in _batch_setitems
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 331, in save
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 419, in save_reduce
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 286, in save
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 649, in save_dict
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 681, in _batch_setitems
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 331, in save
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 419, in save_reduce
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 286, in save
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 649, in save_dict
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 681, in _batch_setitems
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 331, in save
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 396, in save_reduce
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 286, in save
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 748, in save_global
PicklingError: Can't pickle <type 'thread.lock'>: it's not found as thread.lock
Traceback (most recent call last):
  File "<string>", line 37, in <module>
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\multiprocessing", line 145, in freeze_support
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\multiprocessing.forking", line 337, in freeze_support
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\multiprocessing.forking", line 381, in main
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 1378, in load
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 858, in load
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pickle", line 880, in load_eof
EOFError

stephendade, 2017-03-27:
Yep, another threading incompatibility issue with Windows (like with MAVExplorer). Hopefully won't be too difficult to fix

stephendade, 2020-08-07:
This is fixed and working as of Aug 2020

closed 2020-08-07 by stephendade