ArduPilot/MAVProxy #402: Waypoints are cyclically re-requested by copter on SITL

khancyr, 2017-04-07:
See http://discuss.ardupilot.org/t/waypoints-are-cyclically-re-requested-by-copter-on-sitl/16200
It append when Mavproxy is used as a proxy.

To reproduce the bug, use launch SITL. Connect APMplanner or another GCS, create a mission, write the mission and see mavproxy flood with "re-requesting WPs [0, 1, 2, 3, 4]"

Probably related to https://github.com/ArduPilot/MAVProxy/commit/b7765a019224274796c389ad97606a62e463e968 ?

AndKe, 2018-02-23:
are you sure it's only happening in SITL ?
I can surely reproduce it in SITL, but I also seen this in real life.

amilcarlucas, 2018-10-17:
This has been fixed in the latest version

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

efischer91, 2019-05-22:
I'm still having this problem with the current release. I start up mavproxy and SITL with `sim_vehicle.py -L Ballarat --map --console -D -G` and after I connect to it with Dronekit (`connect(127.0.0.1:14551)`), the error "re-requesting WPs [0, 1, 2, 3, 4]" appears.

amilcarlucas, 2019-05-23:
Make sure you use the latest 2.9.2 dronekit, together with the latest pymavlink. That bug has been fixed already.

efischer91, 2019-05-23:
I'm using dronekit 2.9.2 and pymavlink 2.3.4 and I still get that error though.

Heyuchen-330, 2021-09-21:
I have the same problem after starting mavros by`roslaunch mavros apm.launch fcu_url:=udp://127.0.0.1:14550@`

Heyuchen-330, 2021-09-21:
> I have the same problem after starting mavros by`roslaunch mavros apm.launch fcu_url:=udp://127.0.0.1:14550@`

solved by clearing the mission

timtuxworth, 2022-05-31:
I'm getting this on master/plane as at 2022-May-31

rmackay9, 2022-05-31:
The cause is probably different than the original report but I've re-opened this because I'm seeing it as well.

timtuxworth, 2022-06-01:
`wp list` made it go away (for now)

timtuxworth, 2022-06-02:
I just got it again. I had a clean clone (4.1), started SITL, all good, did a takeoff (crashed), switched to manual mode and started getting these messages. There was no mission or fence on the plane, it should have been a brand new, empty eeprom.bin,

timtuxworth, 2022-06-02:
Oh. Ok so I did this:
1. Deleted the eeprom.bin
2. Reloaded params from a param file (I had downloaded this from Tridge's definition for the AeroScout)
3. Entered mavlink command: takeoff 50
4. Entered mavlink command: arm throttle

So is it possible there is something in the param file I loaded that causes the problem? This is it here:
https://www.dropbox.com/s/6p8t8qrr1hh3b1d/aeroscout.param?dl=0

I got it from here:
https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/Tridge/Planes/Aeroscout

MONEEBFAMOUS, 2022-08-05:
> `wp list` made it go away (for now)

Kindly explain what did you mean by doing "wp list" made it go away? can you elaborate what you did to stop the flooding of wps re-requesting in terminal,

dvelozjr, 2024-08-14:
I also cleared the mission in MAVProxy, after "WP: mission received" message in MAVROS, by issuing this command: rosservice call /mavros/mission/clear "{}"